Reedyuk/blue-falcon #16: Android - Stop scanning does not work & potential fix

kirandwarkan, 2019-10-08:
Hi, 

I noticed that the `stopScanning()` implementation for Android does not actually stop scanning for devices.
However, adding the method call
 
`bluetoothManager.adapter?.bluetoothLeScanner?.stopScan(mBluetoothScanCallBack)`

to the Android implementation of the `stopScanning` method works.
Is there a reason why this call is not being made in that method? I have only tested this in my app, but wouldn't mind doing a pull request for it. Being new to Bluetooth, I'm not 100% sure what else I would need to test to make sure the potential fix is ok.

Reedyuk, 2019-10-09:
Possibly a bug, let me investigate but I thought I had put in a similar line of code for the stop scanning method.

Reedyuk, 2019-10-09:
I have pushed those changes to master, i will publish on nexus when i can.

kirandwarkan, 2019-10-09:
Thanks for the quick response. I'm using the project relatively off the latest master since I needed the serviceUUID changes. I'll try using 0.5.0 from nexus.

However, I can't see that stop scanning code changes on master. Perhaps forgot to push to github?

Reedyuk, 2019-10-09:
Not sure how that happened, please pull master now, its there.

Reedyuk, 2019-10-09:
Also could you reply if this is working as expected for you so i can close this issue.
If you decide to use this library in an app, it would be great to add you to the list of Apps using this library

kirandwarkan, 2019-10-09:
Tried latest master and that works.
As for the app, I'm still in the process of building it and ensuring that I can meet the requirements for our BLE functionality. Will let you know if we settle on using blue-falcon and release the app on the Play Store.

closed 2019-10-09 by kirandwarkan