Reedyuk/blue-falcon #24: iOS build issues

felislynx-silae, 2020-02-18:
If i include blue falcon in my multiplatform project in section iosMain, adding
```
implementation("dev.bluefalcon:library-iosx64:0.5.4")
implementation("dev.bluefalcon:library-iosarm64:0.5.4")
```
Results in (while building from Xcode) error

```
* What went wrong:
Could not determine the dependencies of task ':knlib:linkDebugFrameworkIos'.
> Could not resolve all task dependencies for configuration ':knlib:iosCompileKlibraries'.
   > Could not resolve dev.bluefalcon:library-iosarm64:0.5.4.
     Required by:
         project :knlib
      > Unable to find a matching variant of dev.bluefalcon:library-iosarm64:0.5.4:
          - Variant 'iosArm64-api' capability dev.bluefalcon:library-iosarm64:0.5.4:
              - Incompatible attribute:
                  - Required org.jetbrains.kotlin.native.target 'ios_x64' and found incompatible value 'ios_arm64'.
              - Other attributes:
                  - Found org.gradle.status 'release' but wasn't required.
                  - Required org.gradle.usage 'kotlin-api' and found compatible value 'kotlin-api'.
                  - Required org.jetbrains.kotlin.platform.type 'native' and found compatible value 'native'.
          - Variant 'metadata-api' capability dev.bluefalcon:library-iosarm64:0.5.4:
              - Incompatible attribute:
                  - Required org.jetbrains.kotlin.platform.type 'native' and found incompatible value 'common'.
              - Other attributes:
                  - Found org.gradle.status 'release' but wasn't required.
                  - Required org.gradle.usage 'kotlin-api' and found compatible value 'kotlin-api'.
                  - Required org.jetbrains.kotlin.native.target 'ios_x64' but no value provided.
```

Providing just
```
implementation("dev.bluefalcon:library-iosx64:0.5.4") or implementation("dev.bluefalcon:library-ios:0.5.1”)
```
Results with:
```
> Task :knlib:cinteropFirebaseCoreIos FAILED
w: skipping /Users/lynx/.gradle/caches/modules-2/files-2.1/dev.bluefalcon/library-iosx64/0.5.4/b8e445b4bb7b3fa076afa9d68c3619fbb226f7fa/library.klib. The abi versions don't match. Expected '[17]', found '14'
w: The compiler versions don't match either. Expected '[]', found '1.3.50-release-11850'
e: Could not find "/Users/lynx/.gradle/caches/modules-2/files-2.1/dev.bluefalcon/library-iosx64/0.5.4/b8e445b4bb7b3fa076afa9d68c3619fbb226f7fa/library.klib" in [/Users/lynx/Workspaces/ios/KNProject/Pods, /Users/lynx/.konan/klib, /Users/lynx/.konan/kotlin-native-macos-1.3.61/klib/common, /Users/lynx/.konan/kotlin-native-macos-1.3.61/klib/platform/ios_x64].

FAILURE: Build failed with an exception.

```

Also there is no "library-ios:0.5.4" when building with jcenter, only 0.5.1 version.

Reedyuk, 2020-02-18:
Hi @felislynx-silae - i think the issue you are getting is because the library was compiled kotlin 1.3.50 and not your version 1.3.61.
I will need to re-compile and upload a new version for you, for now, you could pull the repo locally, update the kotlin version and then publish locally and you should be able to fetch it then.

Jcenter:
The old version of the bluefalcon library was using library-ios, the new version is using the names with their architecture - library-iosx64, library-iosarm64

felislynx-silae, 2020-02-18:
Ok, thank you for response.
Is this possible for you, to publish 1.3.61 compiled this week? If no, then i'll pull the repo locally (usually i try to avoid pulling too much libs as local repos :) ) and update to jcenter when you upload new version.

Reedyuk, 2020-02-18:
i will try to do it this week.

Reedyuk, 2020-02-19:
I have just released version 0.5.5, please give this a few hours to display in maven central.
This will support 1.3.61. Re-open this issue if you still have the same problem.

closed 2020-02-19 by Reedyuk