Reedyuk/blue-falcon #26: Help with ios BluetoothUnknownException

felislynx-silae, 2020-02-19:
First, thank you for upgrading library to latest Kotlin version.
Now i'm trying to use it with iphone.
I've empty project, added
Privacy - Bluetooth Always Usage Description
Privacy - Bluetooth Peripheral Usage Description
into Info.plist
and after running scan i've got
BluetoothUnknownException - Unknown error happened
Initialization of library is done like in sample: LibraryBlueFalcon(context: UIView(), serviceUUID: nil)
To be honest, i've no idea how to debug it or what is wrong with it :)

Reedyuk, 2020-02-19:
Are you running this in the simulator? Any bluetooth services cannot be used on the simulator.

felislynx-silae, 2020-02-19:
I've checked this on 5s and 11 (physical devices)

Reedyuk, 2020-02-19:
Can you send me a zipped version of the project? I would also suggest you have a look at the example projects in this repo.

felislynx-silae, 2020-02-19:
By the way, in my code its named LibraryBlueFalcon because kotlin made header file with prefixes to distinguish K/N libraries

closed 2020-02-19 by Reedyuk