Reedyuk/blue-falcon #27: Lack of TRANSPORT_LE for Android prevents from working with BLE UART connections

felislynx-silae, 2020-02-20:
In Android when you use connectGatt you can pass TRANSPORT_LE as last parameter which might be required for some devices. Same thing can be done via changing characteristic descriptor parameter via gatt. Currently if i create gatt uart server on Raspberry Pi, i can connect to it and discover services/characteristics but read/writes are not working. This is not a case for iOS which works.

**Describe the solution you'd like**
Add parameter to enable TRANSPORT_LE or expose descriptor change for characteristic

**Additional context**
https://scribles.net/creating-ble-gatt-server-uart-service-on-raspberry-pi/ <-- sample code for raspberry pi 3
https://stackoverflow.com/questions/46273499/trouble-connecting-android-app-to-raspberry-pi-3-over-ble <-- some explanation

Reedyuk, 2020-02-20:
I will try to have a look at this in the coming weeks, otherwise you are welcome to add your own PR with the changes.

Reedyuk, 2020-04-01:
Hi @felislynx-silae, i am just working on the change now, you will be able to modify a property on the android BlueFalcon object called transport method, i will adjust the android example to accommodate for this.
[image]

closed 2020-04-01 by Reedyuk