Reedyuk/blue-falcon #31: Characteristic indication

virulentHive, 2020-04-27:
Would it be possible to add characteristic indication capability beside notification? Our specific usecase and GATT server device requires that we set indication on a characteristic.

Reedyuk, 2020-04-27:
So you would want to send an enum something like:
    public static final byte[] ENABLE_NOTIFICATION_VALUE
    public static final byte[] ENABLE_INDICATION_VALUE
    public static final byte[] DISABLE_NOTIFICATION_VALUE

Would you also want to be able to set the descriptor value too?

Just trying to think how i can make this as generic as possible to support Android, iOS, MacOS.

virulentHive, 2020-04-27:
Wow that was fast, thanks. :)

Personally I think it would be best to provide an indicateCharacteristic() method (with the same arguments as its notify counterpart). It could then set the ENABLE_INDICATION_VALUE in the descriptor internally. No other descriptor value setting is needed I think.
AFAIK on iOS it wouldn't matter as it handles it automatically. No clue about macOS though but I imagine it works the same.

Reedyuk, 2020-04-27:
Yep makes sense, let me see if i can squeeze out a change over the weekend, otherwise, happy for you to fork and put a pr in.

virulentHive, 2020-04-27:
Just published a PR. I also added the possibility to set both notification and indication. That makes it more in line with iOS functionality.

Reedyuk, 2020-04-27:
I have merged and published as 0.6.5, this should appear in maven central within the next few hours.

closed 2020-04-27 by Reedyuk