Reedyuk/blue-falcon #33: Characteristic value string encoded with UTF-8 alters byte values

virulentHive, 2020-04-28:
Currently the BT characteristic value is represented as a String, which is encoded from byte[] (Android) and NSData (iOS) with UTF-8 encoding. This is OK if the characteristic value contains byte values in the UTF-8 charset but causes problems otherwise (replaces unknown byte value with the Unicode replacement character U+FFFD "�").

Example:
```
Characteristic value byte[] sent by GATT server: 102,-120,51,64
Characteristic value string encoded from the byte[]: f�3@
Characteristic value byte[] decoded from the string: 102,-17,-65,-67,51,64
```
In this specific case our device is sending 4 byte packets with little endian ordering. The value -120 is invalid in UTF-8.

I think it would be better to use a ByteArray as the characteristic value type. Then everyone can decide in their client code if it will be converted to String or used as a ByteArray.

Would this cause any problems? Does using String as value have other reasons I'm not seeing?
If this works I'm more than happy to submit a PR.

Thanks!

Reedyuk, 2020-04-28:
No, thats my bad, i was lazy and with my custom bluetooth device i was sending over string. Happy for you to put a PR in. Cheers.

virulentHive, 2020-04-28:
Oh, right. :)
Submitted the PR.

closed 2020-04-28 by Reedyuk

virulentHive, 2020-04-29:
Thanks for the update.
The new version is not on maven central though. The gradle.properties shows the previous version number.

Reedyuk, 2020-04-29:
strange, let me do it again now.