Reedyuk/blue-falcon #35: Characteristic value changed to ByteArrray but writeCharacteristic() is still using String

PeterKucera, 2020-05-13:
Hi all,

I am using latest version of library ("dev.bluefalcon:library:0.6.6"), 

and even after [this fix](https://github.com/Reedyuk/blue-falcon/commit/800d23c1ba2a1c777d10148ede368bf82b45062f)

method writeCharacteristic() still accepts only String argument. 

To achieve bidirectional communication only using bytes it's neccessary to either change argument type of writeCharacteristic(value:String) to writeCharacteristic(value:ByteArray) or provide dedicated method for writing raw bytes.

Reedyuk, 2020-05-13:
Yes makes sense, i will see if i can change this for you or if you wish to put a PR in, i can review and merge in.

felislynx-silae, 2020-06-08:
@Reedyuk i've made quick patch https://drive.google.com/file/d/13VH37vWAkf2DwVGbGIrxBhRbst6n2q-9/view?usp=sharing

With Android implementation, i have no idea how to implement this in iOS/macos.
But i hope this might be a good start. I also belive this feature is quite important

Reedyuk, 2020-06-08:
I cant access it, if you want to just fork the repo and put a PR in against my master, then i can add the ios/macos changes too.

felislynx-silae, 2020-06-08:
@Reedyuk sorry, new google sharing settings are annoying, i've made correction and now you should be able to access them. This is very fast draft (i think it is correct), i'm now under huge deadline :D so patch is best i could do in free time

Reedyuk, 2020-06-08:
No worries, thanks for the update, when i get a moment i will try it out and add the relevant ios changes.
I will keep this issue open until its merged in.
Can i ask, are you using Blue-Falcon in a production app? would be great to see who is using it.

felislynx-silae, 2020-06-08:
Yes, i'm using it in production apps. Not yet released. I'm working on both iOS and Android version of app :). When it will be released i'll let you know with a link if you want

Reedyuk, 2020-06-08:
Excellent, yes a link would be great when you release.
I'm hoping that jetbrains fixes the threading issues on iOS with a new version of kotlin coroutines, once this is in place, we will be able to use multithreading correctly.

closed 2020-06-09 by Reedyuk

Reedyuk, 2020-06-10:
https://search.maven.org/artifact/dev.bluefalcon/library/0.6.7/pom
Version 0.6.7 released and in maven central

felislynx-silae, 2020-06-15:
@Reedyuk in maven central there is released 0.6.7 version for every platform except for Android :)

Reedyuk, 2020-06-15:
ok, i have just pushed 0.6.8, this should appear in the next few hours. This one definitely has android, im not sure what happened there.

PR #36 (2020-06-09): 35 bytearray write
Fixes #35 
Thanks to @felislynx-silae