Reedyuk/blue-falcon #37: [ANDROID] OnConnected called when device is disconnected

felislynx-silae, 2020-06-18:
This code listens to connection change event, it doesn't check what's new state, it always tries to "connect" to device and sends that information out. So if i'm connected and turn off external device, i get information that i'm connected :)
Didn't checked iOS (will do in upcoming weeks) yet.
```
override fun onConnectionStateChange(gatt: BluetoothGatt?, status: Int, newState: Int) {
            super.onConnectionStateChange(gatt, status, newState)
            log("onConnectionStateChange")
            gatt?.let { bluetoothGatt ->
                bluetoothGatt.device.let {
                    addGatt(bluetoothGatt)
                    bluetoothGatt.readRemoteRssi()
                    bluetoothGatt.discoverServices()
                    delegates.forEach {
                        it.didConnect(BluetoothPeripheral(bluetoothGatt.device))
                    }
                }
            }
        }
```

Reedyuk, 2020-06-18:
good spot, we should check the status/new state to determine if connected.

Reedyuk, 2020-06-20:
Hi @felislynx-silae - do you want to check my PR, this should address the issue you were experiencing.

felislynx-silae, 2020-06-20:
Hi, it looks good. If more comes out during tests i'll  let you know :)

PR #40 (2020-06-21): Added fix for android and connection state change
fixes #37 

Thanks felislynx-silae for reporting the issue.
closed 2020-06-21 by Reedyuk