Reedyuk/blue-falcon #38: [Android] Exposing autoConnect flag

PeterKucera, 2020-06-18:
Hi all,

In [BlueFalcon Android actual class](https://github.com/Reedyuk/blue-falcon/blob/master/library/src/androidMain/kotlin/dev/bluefalcon/BlueFalcon.kt) , method connectGatt is called with flag autoconnect set to false by default. 

I understand this is for a reason - in iOS there's no such thing as autoConnect flag (autoConnect on iOS is implicit - correct me if I'm wrong).

But IMO there is still need to expose this autoConnect flag to the developer, so connection initiated by Android stack will be possible, when device comes into range. 

So my sugestion would be - add optional parameter to connect method, e.g. 

connect(bluetoothPeripheral: BluetoothPeripheral, autoConnect:Boolean = false).

On platforms that do not offer setting explicit autoConnect flag(such as iOS, MacOS), this argument would be ignored.

Any comments / different ideas how to approach this ? 

Thanks

Reedyuk, 2020-06-18:
Yes sounds like a logical approach. Do you want to put a PR in for it.

felislynx-silae, 2020-06-18:
Good point. Worth to remember that there was a bug in autoconnect feature (2-3 years ago). Google is not well known with bugfixing in some Android internals ;). Is it working correctly now?

Reedyuk, 2020-06-21:
https://github.com/Reedyuk/blue-falcon/pull/42

PR #42 (2020-06-21): Added auto connect property.
fixes #38
closed 2020-06-21 by Reedyuk