Reedyuk/blue-falcon #39: Characteristic updates after device reconnection

PeterKucera, 2020-06-18:
Hi all,

I am currently experiencing following issue:

BlueFalcon version -  0.6.6
Device - Samsung Galaxy XCover 4, Android 9

**Expected:**

1. Init delegate class for BLE device ✅
2. Connect to peripherial (successful connection, characteristic discovery and call notifyCharacteristic) ✅
3. writeCharacteristic & didCharacteristicValueChanged methods work as expected ✅
4. Disconnect from peripherial ✅
5. Connect to peripherial (successful connection, characteristic discovery and notifyCharacteristic call) ✅
6. writeCharacteristic & didCharacteristicValueChanged methods work as expected ✅

**Actual:**

1. Init delegate class for BLE device ✅
2. Connect to peripherial (successful connection, characteristic discovery and call notifyCharacteristic) ✅
3. writeCharacteristic & didCharacteristicValueChanged methods work as expected✅
4. Disconnect from peripherial ✅
5. Connect to peripherial (successful connection, characteristic discovery and notifyCharacteristic call) ✅
6. writeCharacteristic & didCharacteristicValueChanged methods DO NOT work as expected: writeCharacteristic is called but has no effect, didCharacteristicValueChanged is not called at all - in conclusion no data are passed back & forth ❌

Looks like after device reconnection, there  is some state held by BleAndroid / BlueFalcon library that prevents the app from sending/receiving updates. When I reinitialize BleAndroid object, all starts to work as expected again.

Anyone else have noticed problems after device reconnection ?

felislynx-silae, 2020-06-18:
If using few characteristics with notification on, try placing delay 350ms between each enableNotification call. I had this issue, and placing delay solved my problem

Reedyuk, 2020-06-18:
as @felislynx-silae - i think i did something similar, adding an artificial delay. I think it could be a bug with Android.
I would need to investigate further to understand the root cause.

kazyh, 2020-07-14:
Maybe this is somehow connected with issue I reported #45

Reedyuk, 2020-08-01:
@felislynx-silae - can you try again with 0.7.2 library and confirm if your issue still exists?

felislynx-silae, 2020-08-10:
@Reedyuk which issue? If you mean enableNotification one, i solved it by adding delay weeks ago :)

closed 2020-08-11 by Reedyuk

PeterKucera, 2020-08-12:
I have managed to resolve this issue by using autoConnect flag. Using autoConnect method connect() is called only once and works as expected.