Reedyuk/blue-falcon #43: iOS Error while requested connection to device

felislynx-silae, 2020-06-24:
2020-06-24 07:56:19.182475+0200 App[6530:4475232] [CoreBluetooth] API MISUSE: Discovering characteristic descriptors for peripheral <CBPeripheral: 0x2802d0960, identifier = 2BF80158-F03A-AEEE-0FD4-598964BCA682, name = RPIDevice, state = connected> while delegate is either nil or does not implement peripheral:didDiscoverDescriptorsForCharacteristic:error:

Reedyuk, 2020-06-24:
yeah i need to fix that, simple enough i hope.

felislynx-silae, 2020-06-24:
I hope so :D i was happy that all roadblocks were cleared and this happened :(

felislynx-silae, 2020-06-25:
I did little research (couldn't check since attaching BlueFalcon into my Kotlin Multiplatform project made ios build to not see Blue Falcon classes :/ ). It looks like delegate on peripheral is set too late (currently in didConnect). Shouldn't we set it before calling connect?

felislynx-silae, 2020-06-25:
Did little research:
    `@kotlinx.cinterop.ObjCMethod public open fun peripheral(peripheral: platform.CoreBluetooth.CBPeripheral, didDiscoverDescriptorsForCharacteristic: platform.CoreBluetooth.CBCharacteristic, error: platform.Foundation.NSError?): kotlin.Unit { /* compiled code */ }` 
Is present in knm, but we are unable to do override since those two have same signature and overloads are conflicting.

```
override fun peripheral(
            peripheral: CBPeripheral,
            didDiscoverDescriptorsForCharacteristic: CBCharacteristic,
            error: NSError?
        ) {
            
        }

        override fun peripheral(
            peripheral: CBPeripheral,
            didUpdateValueForCharacteristic: CBCharacteristic,
            error: NSError?
        ) {}
```

Reedyuk, 2020-06-28:
Hi @felislynx-silae - here is a PR for the fix to stop the error message of misuse api usage.
https://github.com/Reedyuk/blue-falcon/pull/44/files

felislynx-silae, 2020-06-29:
Hi @Reedyuk I've checked your PR with devices and it is connecting and communicating. So i think, ready to merge :)

closed 2020-06-29 by Reedyuk

Reedyuk, 2020-06-29:
released 0.7.0

PR #44 (2020-06-29): Added fix for ios/macos with overloading methods.
fixes #43