Reedyuk/blue-falcon #45: Android not calling close() method on BluetoothGatt when disconnecting

kazyh, 2020-07-14:
It is propably only android issue

When calling connect and disconnect device several times during one app session GattCallbacks are aggregated and I get multiplied callbacks about characteristic changed. 

This happens because in BluetoothGatt.connect (it is called by BluetoothDevice.connectGatt in BlueFalcon.connect) callback is registered. But BlueFalcon.disconnect method calling only BluetoothGatt.disconnect and should also call BluetoothGatt.close (in close() method callback is unregistered).

kazyh, 2020-07-14:
It will also solve problem with double didDisconnect call (didDisconnect is called direct from bluefalcon disconnect method and it is called again when gattCallbeck.onConnectionStateChange notify about newState == STATE_DISCONNECTED ) .

If BluetoothGatt.close() would be called in disconnect method it will prevent from any callbacks from disconnected device.

Reedyuk, 2020-07-14:
ok sounds logical, i will look into it. - PR's always welcome.

Reedyuk, 2020-08-01:
Issue has been closed, please checkout release 0.7.2

PR #47 (2020-08-01): Added close when user wishes to disconnect.
Fixes #45
closed 2020-08-01 by Reedyuk