Reedyuk/blue-falcon #46: Library metadata is missing for kotlin MP

xPutnikx, 2020-07-27:
Hi,
 
I'm trying to use the latest version of the library (0.7.0 and 0.7.1) and experience some problems with the dependencies in Kotlin multiplatform project.
I was able to reproduce the issue using the Kotlin-MP sample.

## The issue
When using the latest version of the library in the kotlin-multiplatform project, the dependencies in `commonMain` can't be resolved.

## Steps to reproduce
1. Open the Kotlin-MP sample from this repository.
2. Sync the project
3. Open commonMain -> DevicesViewModel.kt
4. Check that all dependencies resolved
5. Change BlueFalcon version from 0.6.6 to 0.7.1 (in gradle.properties)
6. Open commonMain -> DevicesViewModel.kt

### Expected behavior
Dependencies remain resolved

### Actual behavior
dev.bluefalcon can't be found.
[image]

## Notes
I have done some research around the issue and compared dependencies for both versions (0.6.6 which works and 0.7.0-0.7.1 which doesn't)

In 0.6.6 version you have a package `library-metadata`
[image]
**0.6.6**

In 0.7.1 it disappered
[image]
**0.7.1**

I guess it might be some issue with packaging the library, but not sure. Would be glad to share any additional info needed to help to solve the issue.

Reedyuk, 2020-07-27:
Yes, I must have broken it when i moved over to using Github actions. I will need to investigate why its not packaging correctly.
Thanks

xPutnikx, 2020-07-27:
I have done a bit of investigation and tests just now. 
I see an issue on Kotlin tracker which might be related to that https://youtrack.jetbrains.com/issue/KT-34038

It looks like only AndroidStudio fails to resolve dependencies while the compiler actually works and compiles the project properly.

Reedyuk, 2020-08-01:
What version of gradle are you using? im finding at the moment version 6+ has alot of issues.
I have reverted back to 5.6.4

Reedyuk, 2020-08-01:
So checked to make sure maven central has all the libraries, it does:
[image]

So im going to close this issue, as i think this is unfortunately a kotlin issue.

closed 2020-08-01 by Reedyuk