Reedyuk/blue-falcon #49: Unresolved reference - BlueFalcon

PeterKucera, 2020-08-13:
Hi all,

I am experiencing following issue:

Inside commonMain package Android Studio reports that import dev.bluefalcon.* cannot be resolved, but when I try to build & run project - the app builds and works as expected. Only problem is that all BlueFalcon related methods / imports are not resolved and code readability is bad, since all unresolved objects are highlited in red.

Anyone else experienced this issue ?

Which version of Kotlin/ BlueFalcon / Android Studio do you use  ?

I'm using: 
Android Studio 3.6.1
Kotlin version 1.3.71 
BlueFalcon 0.7.2

Reedyuk, 2020-09-16:
This was a reported issue with Kotlin, I would recommend you upgrade to a newer version.

marcforn, 2020-09-27:
Hi all, 

I'm having exactly the same issue. Impossible to resolve BlueFalcon reference.

I'm using:
Android Studio 4.1 RC3
Kotlin version 1.4.10
BlueFalcon 0.7.2

Reedyuk, 2020-09-30:
I have just uploaded version 0.8.0, should be there in a few hours, please can you try and see if it works ok.

marcforn, 2020-10-01:
I do confirm. With newest version imports are resolved successfully.

closed 2020-10-24 by Reedyuk