Reedyuk/blue-falcon #50: Writing characteristic not working after reconnect.

kazyh, 2020-08-24:
Issue created with #45 fix. 
After disconnect by calling disconnect method and reconnect again with device within same BlueFalcon instance, we won’t be able to write to gatt characteristic.

It is because when we close the gatt connection in disconnect method, the method onConnectionStateChange from BluetoothGattCallback will never be called and as a result removeGatt method won’t be called. Thus the item in the Gatt list is not removed. In such cases when trying to write a characteristic we take an old object which is not currently connected to the device.

In my opinion there are two solutions
add additional call of removeGatt in disconnect method
moving close() method from disconnect to onConnectionStateChange in BluetoothGattCallback so gatt will be closed after remove. 

First solution should work properly but it is not that clean and neat.
Second is better but You need to check if moving close method won’t be an issue in disconnect scenarios and In that case You need to remove delegates.forEach { it.didDisconnect(bluetoothPeripheral) } from disconnect method to avoid calling it twice.

Reedyuk, 2020-08-24:
hmm, second would be better. I think we should try it and see if it doesnt break anything.

dishank-betaflux, 2022-09-24:
Any updates on it?

Reedyuk, 2022-09-24:
No update yet, but if you want to put a pr in, would happily review it