Reedyuk/blue-falcon #51: Method didDiscoverDevice returns Bluetooth MAC address instead of name

PeterKucera, 2020-08-26:
**Steps to reproduce**

In method didDiscoverDevice,  bluetoothPeripherial.name contains instead of string value with device name only MAC address of that given device.

Expected 
I/BlueFalcon: Discovered device : DeviceNameXY

Actual
I/BlueFalcon: Discovered device : 46:95:8F:86:74:A0

**Additional info**
Device - Samsung XCover Pro
OS version - Android 10
Library version - "dev.bluefalcon:library:0.7.2"

Anyone else experienced similar issue before? 

Note:
This could be also an Android issue, because sometimes when I am trying to pair a new device in a place where are too many of advertising Bluetooth devices at the same time, even default system settings app couldn't read device names, and shows just MAC addresses.

PeterKucera, 2020-08-26:
Now I see what causes this behaviour, if device name is null, device address is returned instead,

From class BluetoothPeripheral:

` actual val name: String?
        get() = bluetoothDevice.name ?: bluetoothDevice.address`

Reedyuk, 2020-08-26:
Yep its intentional to ensure something is returned, as apposed to nothing.

closed 2020-08-26 by Reedyuk