Reedyuk/blue-falcon #53: GATT connection not closing (ANDROID)

felislynx-silae, 2020-09-16:
Hi, during more stress test of my app, i've discovered that Android device after some time causes peripheral to stop accept connections and stops from being visible in ble scan. This is not a case for iOS.
Steps to reproduce:

0. Transport type LE
1. Start LE scan if peripheral MAC found, connect
2. If onConnected emited, stop LE scan
3. acquire all characteristics (setup notifications etc)
4. wait a little time
5. disconnect
6. wait a little time
7. repeat from point 1

After few loops, you will not be able to find peripheral MAC. Reseting peripheral or Android BT restores connectivity, removing Initializing BlueFalcon again also solves this issue. As far as i can tell, the GATT connection is held and this causes to reach maximum available connections. Good peripheral to test is Raspberry PI 3b+ with simple code using bluez.

Reedyuk, 2020-09-16:
Nice catch, so maybe we need to purge the Gatt objects?

felislynx-silae, 2020-09-16:
I would start with setting references to null. Calling close and disconnect or searching for more solutions. This is clearly issue with android bt stack since internet is full of posts about ble issues :) shame that we need to do something with this not google. Kudos for Apple for better BT stack. If you provide some branches to test i will do stress tests ASAP (between 5am and 2pm :) )

PeterKucera, 2020-09-17:
I can reproduce the exact same issue. 

I have tried to investigate a little bit, e.g. have a look on [following link](https://stackoverflow.com/questions/35986549/totally-disconnect-a-bluetooth-low-energy-device) .

"As previously stated, the problem is caused by calling connectGatt multiple times. Each one of those calls create a new BluetoothGatt instance and they are all kept alive, while you have only the last one. Given the fact that sometimes it is needed to call connectGatt multiple times, I just keep all the instances that it returns and call disconnect/close on all of them when I'm done. "

Possible solution for this could be storing all BluetoothGatt objects returned by method connectGatt() in a list and on device disconnection, try to trigger disconnect() + close() on all of these BluetoothGatt objects, not only on one that is returned by mGattClientCallback.gattForDevice() .

felislynx-silae, 2020-09-18:
I think that solution from @PeterKucera might work, worth a try

PeterKucera, 2020-09-21:
Hi all,

I have done some tests and tried to modify the library. 

It looks like approach mentioned in my [previous comment](https://github.com/Reedyuk/blue-falcon/issues/53#issuecomment-694364096) will not fix the issue.

However, it looks like the issue could be calling calling close() method right after disconnect() method - when I added some delay between these 2 method calls, I cannot reproduce the issues I previously had, and therefore it looks like GATT connection is closed properly. 

Please @felislynx-silae can I ask for retesting using this fix ? 

https://github.com/PeterKucera/blue-falcon/commit/67eb76776e1100e2168f3c176460500d45ee18d4

felislynx-silae, 2020-09-24:
@PeterKucera I've tested your commit. It is not fixing this issue. Currently only working solution is nulling all references to BlueFalcon and recreate object

closed 2026-04-14 by Reedyuk