Reedyuk/blue-falcon #54: Library not working on Kotlin 1.4.x

marcforn, 2020-09-28:
Library only works on Kotlin 1.37.2 but not on version 1.4.x

**Error**
e: Could not find "/Users/xxxxxx/.gradle/caches/modules-2/files-2.1/dev.bluefalcon/library-iosx64/0.7.2/a144ef7bdbb2b135d54d02993a459c6f89e4e643/library.klib" in [/Users/xxxxxx/AndroidStudioProjects/kmm-demo, /Users/xxxxxx/.konan/klib, /Users/marc.forn/.konan/kotlin-native-prebuilt-macos-1.4.10/klib/common, /Users/xxxxxx/.konan/kotlin-native-prebuilt-macos-1.4.10/klib/platform/ios_x64].
Daemon vm is shutting down... The daemon has exited normally or was terminated in response to a user interrupt.

This will be fixed by pointing library to newer version of Kotlin.

Looking forward and update :)

Reedyuk, 2020-09-30:
Hi @marcforn - i have just uploaded a new version to maven, this should appear in a few hours. version: 0.8.0 - now supports 1.4.10

closed 2020-09-30 by Reedyuk

marcforn, 2020-10-01:
Great job!! Thanks