Reedyuk/blue-falcon #55: Support for RaspberryPi

felislynx-silae, 2020-09-30:
Well, i've been thinking about some future projects and one of them is related to Kotlin application on RPI. And i noticed that there is no good Bluetooth library for RPI with Java/Kotlin. What do you think about that?

Reedyuk, 2020-09-30:
Its very Python heavy, but totally agree. Happy to look into it, just been very busy.
The same with support Javascript on the front end would be cool(its possible).

felislynx-silae, 2020-09-30:
True, funny enough i was working on 3 projects on RPI. One with Android Things (shame they ditched project), two on linux and java/kotlin (you can use gpio and i2c without many problems). Right now you can even build ui with flutter on RPI which works very well even for alpha. So there is python'less future for RPI in sobe cases :)

weliem, 2021-03-06:
I wrote a java library for Bluez. I use it on my raspberry.

https://github.com/weliem/blessed-bluez

Reedyuk, 2021-03-06:
To be fair @felislynx-silae - there is a chance i could adjust the platforms to support jvm, android and then this way hopefully would work fine on raspberry pi.
Otherwise for now, i guess you could use @weliem's library.

weliem, 2021-03-07:
@Reedyuk , my library is modelled after CoreBluetooth so it should be straightforward to integrate in BlueFalcon...in case you are interested

Reedyuk, 2021-03-09:
@weliem - i tried to integrate your library into Blue-Falcon for a java target, but im getting some errors(im running on a mac so maybe that could be it).
https://github.com/Reedyuk/blue-falcon/pull/67
```
org.freedesktop.dbus.exceptions.DBusException: Cannot Resolve Session Bus Address: MachineId file can not be found
	at org.freedesktop.dbus.connections.impl.DBusConnection.lambda$determineMachineIdFile$5(DBusConnection.java:320)
	at java.base/java.util.Optional.orElseThrow(Optional.java:408)
	at org.freedesktop.dbus.connections.impl.DBusConnection.determineMachineIdFile(DBusConnection.java:320)
	at org.freedesktop.dbus.connections.impl.DBusConnection.getDbusMachineId(DBusConnection.java:303)
	at org.freedesktop.dbus.connections.impl.DBusConnection.getConnection(DBusConnection.java:157)
	at org.freedesktop.dbus.connections.impl.DBusConnection.getConnection(DBusConnection.java:169)
	at org.freedesktop.dbus.connections.impl.DBusConnection.getConnection(DBusConnection.java:219)
	at org.freedesktop.dbus.connections.impl.DBusConnection.newConnection(DBusConnection.java:198)
	at com.welie.blessed.BluezAdapterProvider.<init>(BluezAdapterProvider.java:23)
	at com.welie.blessed.BluetoothCentralManager.<init>(BluetoothCentralManager.java:218)
	at com.welie.blessed.BluetoothCentralManager.<init>(BluetoothCentralManager.java:214)
	at dev.bluefalcon.BlueFalcon.<init>(BlueFalcon.kt:12)
	at dev.bluefalcon.controller.MainController.<init>(MainController.kt:12)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:490)
	at java.base/java.lang.Class.newInstance(Class.java:584)
	at tornadofx.FXKt.find(FX.kt:434)
	at dev.bluefalcon.view.MainView$$special$$inlined$inject$1.getValue(Component.kt:1341)
	at dev.bluefalcon.view.MainView$$special$$inlined$inject$1.getValue(Component.kt:190)
	at dev.bluefalcon.view.MainView.getCtrl(MainView.kt)
	at dev.bluefalcon.view.MainView.<init>(MainView.kt:10)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:490)
	at java.base/java.lang.Class.newInstance(Class.java:584)
	at tornadofx.FXKt.find(FX.kt:434)
	at tornadofx.FXKt.find$default(FX.kt:423)
	at tornadofx.App.start(App.kt:83)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.lambda$launchApplication1$9(LauncherImpl.java:846)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.lambda$runAndWait$12(PlatformImpl.java:455)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.lambda$runLater$10(PlatformImpl.java:428)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.lambda$runLater$11(PlatformImpl.java:427)
	at javafx.graphics/com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:96)
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
Mar 09, 2021 7:56:01 PM tornadofx.DefaultErrorHandler uncaughtException
SEVERE: Uncaught error
java.lang.NullPointerException: no valid dbusconnection provided
	at java.base/java.util.Objects.requireNonNull(Objects.java:246)
	at com.welie.blessed.BluezSignalHandler.createInstance(BluezSignalHandler.java:32)
	at com.welie.blessed.BluezAdapterProvider.<init>(BluezAdapterProvider.java:32)
	at com.welie.blessed.BluetoothCentralManager.<init>(BluetoothCentralManager.java:218)
	at com.welie.blessed.BluetoothCentralManager.<init>(BluetoothCentralManager.java:214)
	at dev.bluefalcon.BlueFalcon.<init>(BlueFalcon.kt:12)
	at dev.bluefalcon.controller.MainController.<init>(MainController.kt:12)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:490)
	at java.base/java.lang.Class.newInstance(Class.java:584)
	at tornadofx.FXKt.find(FX.kt:434)
	at dev.bluefalcon.view.MainView$$special$$inlined$inject$1.getValue(Component.kt:1341)
	at dev.bluefalcon.view.MainView$$special$$inlined$inject$1.getValue(Component.kt:190)
	at dev.bluefalcon.view.MainView.getCtrl(MainView.kt)
	at dev.bluefalcon.view.MainView.<init>(MainView.kt:10)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
[... cut ...]
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.lambda$launchApplication1$9(LauncherImpl.java:846)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.lambda$runAndWait$12(PlatformImpl.java:455)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.lambda$runLater$10(PlatformImpl.java:428)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.lambda$runLater$11(PlatformImpl.java:427)
	at javafx.graphics/com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:96)
```

Reedyuk, 2021-03-09:
I need to be able to use a Multi-platform java bluetooth library to ensure Java consumers on any platform can use blue-falcon.

weliem, 2021-03-09:
Hi, you are getting that error because you dont have a DBus on your mac. At
startup the code tries to establish a connection with the DBus....which
obviously fails on a Mac...but not on Linux machines

weliem, 2021-03-09:
Oh and my library is not a Java multi platform library. Does such a thing even exit?

weliem, 2021-03-10:
I don't think that will work. Bluecove is totally dead and out of date. The nice thing of Kotlin MP is that you don't need one library to back it all. You can have different implementation for each target. So I thought you would only have to add a 'jvm' target to your project and use a Bluez library for that.....

Reedyuk, 2021-03-13:
@felislynx-silae @weliem - can you both check out https://github.com/Reedyuk/blue-falcon/pull/67
I have created a target version, but i dont have a RPI at hand to test.

If you run the RPI-Example, it will create a tar/zip file and you can run the bin/RPI-Example file, it should work.

Reedyuk, 2021-03-14:
@felislynx-silae rpi support and published.

```implementation 'dev.bluefalcon:blue-falcon-rpi:0.9.3'```

closed 2021-03-14 by Reedyuk