Reedyuk/blue-falcon #57: Cannot build KotlinMP example

ln-12, 2020-10-09:
**Describe the bug**
It is not possible to build the KotlinMP example in XCode.
Error message:

```
The Kotlin source set androidAndroidTestRelease was configured but not added to any Kotlin compilation. You can add a source set to a target's compilation by connecting it with the compilation's default source set using 'dependsOn'.
See https://kotlinlang.org/docs/reference/building-mpp-with-gradle.html#connecting-source-sets
w: Skipping "/Users/lneumann/.gradle/caches/modules-2/files-2.1/org.jetbrains.kotlinx/kotlinx-coroutines-core-iosarm64/1.3.8/ab1f410af27e33b328b0e7b3b75875953822e9eb/kotlinx-coroutines-core.klib" as it is a pre 1.4 library
e: Could not find "/Users/lneumann/.gradle/caches/modules-2/files-2.1/org.jetbrains.kotlinx/kotlinx-coroutines-core-iosarm64/1.3.8/ab1f410af27e33b328b0e7b3b75875953822e9eb/kotlinx-coroutines-core.klib" in [/Users/lneumann/Projekte/SteigtUm/blue-falcon/examples/KotlinMP/iosApp, /Users/lneumann/.konan/klib, /Users/lneumann/.konan/kotlin-native-prebuilt-macos-1.4.10/klib/common, /Users/lneumann/.konan/kotlin-native-prebuilt-macos-1.4.10/klib/platform/ios_arm64].
Daemon vm is shutting down... The daemon has exited normally or was terminated in response to a user interrupt.
----- End of the daemon log -----

FAILURE: Build failed with an exception.

* What went wrong:
Gradle build daemon disappeared unexpectedly (it may have been killed or may have crashed)

* Try:
Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Exception is:
org.gradle.launcher.daemon.client.DaemonDisappearedException: Gradle build daemon disappeared unexpectedly (it may have been killed or may have crashed)
	at org.gradle.launcher.daemon.client.DaemonClient.handleDaemonDisappearance(DaemonClient.java:250)
	at org.gradle.launcher.daemon.client.DaemonClient.monitorBuild(DaemonClient.java:223)
	at org.gradle.launcher.daemon.client.DaemonClient.executeBuild(DaemonClient.java:185)
	at org.gradle.launcher.daemon.client.DaemonClient.execute(DaemonClient.java:148)
	at org.gradle.launcher.daemon.client.DaemonClient.execute(DaemonClient.java:99)
	at org.gradle.launcher.cli.RunBuildAction.run(RunBuildAction.java:56)
	at org.gradle.internal.Actions$RunnableActionAdapter.execute(Actions.java:212)
	at org.gradle.launcher.cli.DefaultCommandLineActionFactory$ParseAndBuildAction.execute(DefaultCommandLineActionFactory.java:294)
	at org.gradle.launcher.cli.DefaultCommandLineActionFactory$ParseAndBuildAction.execute(DefaultCommandLineActionFactory.java:266)
	at org.gradle.launcher.cli.DebugLoggerWarningAction.execute(DebugLoggerWarningAction.java:82)
	at org.gradle.launcher.cli.DebugLoggerWarningAction.execute(DebugLoggerWarningAction.java:30)
	at org.gradle.launcher.cli.WelcomeMessageAction.execute(WelcomeMessageAction.java:92)
	at org.gradle.launcher.cli.WelcomeMessageAction.execute(WelcomeMessageAction.java:38)
	at org.gradle.launcher.cli.NativeServicesInitializingAction.execute(NativeServicesInitializingAction.java:44)
	at org.gradle.launcher.cli.NativeServicesInitializingAction.execute(NativeServicesInitializingAction.java:26)
	at org.gradle.launcher.cli.ExceptionReportingAction.execute(ExceptionReportingAction.java:41)
	at org.gradle.launcher.cli.ExceptionReportingAction.execute(ExceptionReportingAction.java:26)
	at org.gradle.launcher.cli.DefaultCommandLineActionFactory$WithLogging.execute(DefaultCommandLineActionFactory.java:259)
	at org.gradle.launcher.Main.doAction(Main.java:35)
	at org.gradle.launcher.bootstrap.EntryPoint.run(EntryPoint.java:50)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.gradle.launcher.bootstrap.ProcessBootstrap.runNoExit(ProcessBootstrap.java:60)
	at org.gradle.launcher.bootstrap.ProcessBootstrap.run(ProcessBootstrap.java:37)
	at org.gradle.launcher.GradleMain.main(GradleMain.java:31)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.gradle.wrapper.BootstrapMainStarter.start(BootstrapMainStarter.java:31)
	at org.gradle.wrapper.WrapperExecutor.execute(WrapperExecutor.java:108)
	at org.gradle.wrapper.GradleWrapperMain.main(GradleWrapperMain.java:63)

* Get more help at https://help.gradle.org

> Task :app:compileKotlinIos
Command PhaseScriptExecution failed with a nonzero exit code
```
**To Reproduce**
Steps to reproduce the behavior:
1. Clone repo
2. Open XCode project inside KotlinMP exmaple
3. Build

**Expected behavior**
Build completes without errors.

**Desktop (please complete the following information):**
 - OS: macOS 10.15.7

**Smartphone (please complete the following information):**
 - Device: iPad Mini
 - OS: iOS 13

Reedyuk, 2020-10-09:
What version of Gradle and also kotlin?

Looks like the coroutines library used is of 1.3.70 and needs to be changed to a 1.4 one.

```e: Could not find "/Users/lneumann/.gradle/caches/modules-2/files-2.1/org.jetbrains.kotlinx/kotlinx-coroutines-core-iosarm64/1.3.8/ab1f410af27e33b328b0e7b3b75875953822e9eb/kotlinx-coroutines-core.klib" in [/Users/lneumann/Projekte/SteigtUm/blue-falcon/examples/KotlinMP/iosApp, /Users/lneumann/.konan/klib, /Users/lneumann/.konan/kotlin-native-prebuilt-macos-1.4.10/klib/common, /Users/lneumann/.konan/kotlin-native-prebuilt-macos-1.4.10/klib/platform/ios_arm64].```

ln-12, 2020-10-09:
Ok, that helped. Now there's another error I can't get rid of. I also had this one in my own project, that's why I wanted to try out the example to see if I'm doing wrong:

```

> Task :app:compileKotlinIos FAILED
e: /Users/<some path>/blue-falcon/examples/KotlinMP/app/src/iosMain/kotlin/sample/scan.kt: (3, 1): @Throws must have non-empty class list

FAILURE: Build failed with an exception.
```

ln-12, 2020-10-09:
If I remove the `@Throws` annotation, Xcode builds the project and runs the app. It seems to work fine at first sight, but in the console XCode outputs: `Unknown error happened`.

Reedyuk, 2020-10-09:
Try kotlin.native.Throws instead?

ln-12, 2020-10-09:
Nope. Either I put the `@Throws` annotation above the method and get `@Throws must have non-empty class list` or I leave it out and get `Function doesn't have or inherit @Throws annotation and thus exception isn't propagated from Kotlin to Objective-C/Swift as NSError.
It is considered unexpected and unhandled instead. Program will be terminated.`

Reedyuk, 2020-10-09:
Sounds very familiar, i recall having a similar issue before and cant remember what i did the fix it. Let me have a think about it.

ln-12, 2020-10-09:
I figured it out now. I can't call `scan()` directly from Swift. Instead, I need to use your `MainCompat` approach like shown [here](https://github.com/Reedyuk/blue-falcon/blob/master/examples/KotlinMP/app/src/commonMain/kotlin/presentation/viewmodels/devices/DevicesViewModel.kt#L24). Maybe you could explicitly state that in the documentation? And I also see that `@UseExperimental` is deprecated and the `NSBluetoothAlwaysUsageDescription` is missing, so I think updating the example would also be a good idea.

closed 2020-10-09 by ln-12

Reedyuk, 2020-10-09:
Ah, yes that was it.
Could you possibly do a PR with the changes? 

Thanks

ln-12, 2020-10-09:
I found that by using the coroutine native-mt dependency we can completely get rid of MainCompat. Everything should be up to date now.