Reedyuk/blue-falcon #60: Add support for new Kotlin IR

ln-12, 2020-10-25:
The library is currently not compatible with [the new Kotlin IR ](https://kotlinlang.org/docs/reference/whatsnew14.html#new-jvm-ir-backend). I setup a minimal example [here](https://github.com/ln-12/kmm_ir_issue). Please add support for this new backend.

Reedyuk, 2020-10-25:
Hi, what sort of impact would this have on people using none kotlin IR projects? I would'nt want to restrict people using this library already.

ln-12, 2020-10-25:
I really don't know. That seems to be a big problem as you can't mix libraries that support and don't support IR. Maybe two artifacts would be appropriate?

Reedyuk, 2020-10-25:
Its currently in alpha, is there a reason for the push to support it right now?

ln-12, 2020-10-25:
You must enable IR if you want to use Jetpack compose.

Reedyuk, 2020-10-25:
I have added a PR to enable IR but i haven't had chance to update android example.

ln-12, 2020-10-25:
Great! Thanks for your fast reply 😄

ln-12, 2021-02-16:
I can conform that it works after updating to Kotlin 1.4.30.

closed 2021-02-16 by ln-12