Reedyuk/blue-falcon #63: changeMTU() method does not work as expected

davidlepilote, 2020-12-17:
When requesting a change of the MTU, the peripheral returned by the callback does not contain services and characteristics. Yet, if you keep a reference of previous characteristics, one cannot write or read on it anymore.

The problem seems to lie in the fact that the MTU change request should appear between the connection and the discovery of services. Yet BlueFalcon abstracts this layer by calling directly the discovery of services as soon as the peripheral is connected.

It could be interesting to add an MTU request within the connect method for instance.

Reedyuk, 2020-12-17:
Hi @davidlepilote - do you want to put a PR in for it and try it?

davidlepilote, 2020-12-17:
I could try yes

closed 2025-02-09 by Reedyuk