Reedyuk/blue-falcon #72: iOS App throws "kotlin.IllegalStateException: There is no event loop. Use runBlocking { ... } to start one."

jamesjmtaylor, 2021-05-30:
**Describe the bug**
After running an XCode project created with KMM I encounter the error "kotlin.IllegalStateException: There is no event loop. Use runBlocking { ... } to start one."

**To Reproduce**
Steps to reproduce the behavior:
1. Launch the compiled iOS application
2. Grant Bluetooth permissions
3. Observe an empty UITableView.

**Expected behavior**
The DevicesViewController should populate with scanned devices.

**Stacktrace**

```
Unknown error happened
kotlin.IllegalStateException: There is no event loop. Use runBlocking { ... } to start one.
    at 0   shared                              0x00000001044fa5f8 kfun:kotlin.Throwable#<init>(kotlin.String?){} + 96
    at 1   shared                              0x00000001044f38f4 kfun:kotlin.Exception#<init>(kotlin.String?){} + 92
    at 2   shared                              0x00000001044f3b64 kfun:kotlin.RuntimeException#<init>(kotlin.String?){} + 92
    at 3   shared                              0x00000001044f3fc0 kfun:kotlin.IllegalStateException#<init>(kotlin.String?){} + 92
    at 4   shared                              0x000000010457a5e0 kfun:kotlinx.coroutines.takeEventLoop#internal + 432
    at 5   shared                              0x000000010457a114 kfun:kotlinx.coroutines.DefaultExecutor#dispatch(kotlin.coroutines.CoroutineContext;kotlinx.coroutines.Runnable){} + 168
    at 6   shared                              0x000000010457b404 kfun:kotlinx.coroutines.NativeMainDispatcher.dispatch#internal + 236
    at 7   shared                              0x000000010456bd04 kfun:kotlinx.coroutines.internal#resumeCancellableWith__at__kotlin.coroutines.Continuation<0:0>(kotlin.Result<0:0>;kotlin.Function1<kotlin.Throwable,kotlin.Unit>?){0§<kotlin.Any?>} + 876
    at 8   shared                              0x0000000104578930 kfun:kotlinx.coroutines.intrinsics#startCoroutineCancellable__at__kotlin.coroutines.SuspendFunction1<0:0,0:1>(0:0;kotlin.coroutines.Continuation<0:1>;kotlin.Function1<kotlin.Throwable,kotlin.Unit>?){0§<kotlin.Any?>;1§<kotlin.Any?>} + 500
    at 9   shared                              0x0000000104578c00 kfun:kotlinx.coroutines.intrinsics#startCoroutineCancellable$default__at__kotlin.coroutines.SuspendFunction1<0:0,0:1>(0:0;kotlin.coroutines.Continuation<0:1>;kotlin.Function1<kotlin.Throwable,kotlin.Unit>?;kotlin.Int){0§<kotlin.Any?>;1§<kotlin.Any?>} + 264
    at 10  shared                              0x000000010454f984 kfun:kotlinx.coroutines.CoroutineStart#invoke(kotlin.coroutines.SuspendFunction1<0:0,0:1>;0:0;kotlin.coroutines.Continuation<0:1>){0§<kotlin.Any?>;1§<kotlin.Any?>} + 420
    at 11  shared                              0x0000000104543350 kfun:kotlinx.coroutines.AbstractCoroutine#start(kotlinx.coroutines.CoroutineStart;0:0;kotlin.coroutines.SuspendFunction1<0:0,1:0>){0§<kotlin.Any?>} + 148
    at 12  shared                              0x0000000104543524 kfun:kotlinx.coroutines#launch__at__kotlinx.coroutines.CoroutineScope(kotlin.coroutines.CoroutineContext;kotlinx.coroutines.CoroutineStart;kotlin.coroutines.SuspendFunction1<kotlinx.coroutines.CoroutineScope,kotlin.Unit>){}kotlinx.coroutines.Job + 436
    at 13  shared                              0x0000000104543794 kfun:kotlinx.coroutines#launch$default__at__kotlinx.coroutines.CoroutineScope(kotlin.coroutines.CoroutineContext?;kotlinx.coroutines.CoroutineStart?;kotlin.coroutines.SuspendFunction1<kotlinx.coroutines.CoroutineScope,kotlin.Unit>;kotlin.Int){}kotlinx.coroutines.Job + 436
    at 14  shared                              0x00000001045815b8 kfun:com.dxfactor.btsdk.presentation.viewmodels.devices.DevicesViewModel#scan(){} + 744
    at 15  shared                              0x00000001045f19d4 objc2kotlin.258 + 132
    at 16  OverlayIOS                          0x000000010419d06c $s10OverlayIOS21DevicesViewControllerC11viewDidLoadyyF + 612
    at 17  OverlayIOS                          0x000000010419d174 $s10OverlayIOS21DevicesViewControllerC11viewDidLoadyyFTo + 32
    at 18  UIKitCore                           0x00000001a47bee50 CC6E5AC7-8248-35F6-8B42-2E25C93DCF0A + 4623952
    at 19  UIKitCore                           0x00000001a47c3408 CC6E5AC7-8248-35F6-8B42-2E25C93DCF0A + 4641800
    at 20  UIKitCore                           0x00000001a470aa94 CC6E5AC7-8248-35F6-8B42-2E25C93DCF0A + 3885716
    at 21  UIKitCore                           0x00000001a470ad9c CC6E5AC7-8248-35F6-8B42-2E25C93DCF0A + 3886492
    at 22  UIKitCore                           0x00000001a470bc60 CC6E5AC7-8248-35F6-8B42-2E25C93DCF0A + 3890272
    at 23  UIKitCore                           0x00000001a470cfe0 CC6E5AC7-8248-35F6-8B42-2E25C93DCF0A + 3895264
    at 24  UIKitCore                           0x00000001a46efe78 CC6E5AC7-8248-35F6-8B42-2E25C93DCF0A + 3776120
    at 25  UIKitCore                           0x00000001a538e398 CC6E5AC7-8248-35F6-8B42-2E25C93DCF0A + 17007512
    at 26  QuartzCore                          0x00000001a5893df4 461501DD-204F-3516-9F5D-BDD0BA19FDA1 + 1400308
    at 27  QuartzCore                          0x00000001a589a398 461501DD-204F-3516-9F5D-BDD0BA19FDA1 + 1426328
    at 28  QuartzCore                          0x00000001a58a56e8 461501DD-204F-3516-9F5D-BDD0BA19FDA1 + 1472232
    at 29  QuartzCore                          0x00000001a57f0d7c 461501DD-204F-3516-9F5D-BDD0BA19FDA1 + 732540
State 5 is .poweredOn
```

**Desktop (please complete the following information):**
 - OS: MacOS 11.4

**Smartphone (please complete the following information):**
 - Device: iPhone X
 - OS: iOS 14.4.2

Reedyuk, 2021-06-03:
Let me have a look and see whats going on

Reedyuk, 2021-06-03:
Hi @jamesjmtaylor - i have updated the KotlinMM example for ios - this is now working, maybe use this as reference.
https://github.com/Reedyuk/blue-falcon/tree/master/examples/KotlinMP-Example
If you are still struggling, then provide me with an example and i can have a look for you.

jamesjmtaylor, 2021-06-03:
@Reedyuk awesome!  Thank you.  I'll give it a go and let you know if I run into any issues.

jamesjmtaylor, 2021-06-03:
@Reedyuk I went through the changes you made at 

https://github.com/Reedyuk/blue-falcon/commit/98dff7bda90f219137ed799f9e25379fc83e3ef7
and
https://github.com/Reedyuk/blue-falcon/commit/afb5b30c8b52fe057fbdaff4dacb69a56f46c739 

Unfortunately I had already implemented these changes prior to your fix (the sample project wouldn't compile otherwise). You can find my source code at https://github.com/jamesjmtaylor/blue-falcon-kmm-sample. After confirming all the changes that you had were present in my project I compiled and ran it in XCode only to receive the same error. Hopefully I'm just overlooking something dumb 😄

jamesjmtaylor, 2021-06-05:
@Reedyuk I found the issue.  Since I'm fairly new to KMM, I was launching the iOS project from XCode, rather than Android Studio.  That meant that the Framework wasn't being recompiled, even after I compiled the Android KMM app to compare this iOS app against.  After making that small change to my workflow it compiled & ran! Thank you for looking into this.

[image]

closed 2021-06-05 by jamesjmtaylor