Reedyuk/blue-falcon #73: Non-null service UUID constructor parameter removes all scan results for iOS

jamesjmtaylor, 2021-06-06:
**Describe the bug**
Non-null service UUID constructor parameter removes all scan results for iOS, even those that match.

**To Reproduce**
Steps to reproduce the behavior:
1. Update the AppDelegate method in the iOS project to pass a service UUID string in the initialization of the BlueFalcon singleton.

**Expected behavior**
Only devices that match that service should be shown.

**Desktop (please complete the following information):**
 - OS: MacOS 11.4

**Smartphone (please complete the following information):**
 - Device: iPhone X
 - OS: iOS 14.4.2

**Additional context**
I recommend the following fix to iosMain/.../BlueFalcon.kt (I would create a branch and open a PR, but I don't have sufficient rights to do so):
[image]

Reedyuk, 2021-06-06:
"Use the forks" or something on those lines.
@jamesjmtaylor - this is exactly what forks were invented, fork this repository, create your changes and when you create a PR you can target the master to being reedyuk/blue-falcon/master.

Good luck!

jamesjmtaylor, 2021-06-06:
@Reedyuk PR is up! https://github.com/Reedyuk/blue-falcon/pull/74

closed 2021-06-06 by Reedyuk