Reedyuk/blue-falcon #75: No such module 'BlueFalcon'

iphonedev4radixweb, 2021-06-29:
opening an example project throw error in DeviceViewModel.swif file

Reedyuk, 2021-06-29:
Which example project? Sounds like its not building the framework, you should have a console error in the build.
Also What version of xcode and kotlin are you building for?

iphonedev4radixweb, 2021-06-29:
example project for iOS . My xcode version is 12.4 .

Reedyuk, 2021-06-29:
Do you have a screenshot or copy of the build output?

iphonedev4radixweb, 2021-06-29:
I am getting build time error . there is no build output
[image]

Reedyuk, 2021-06-29:
Do you have kotlin installed?

Reedyuk, 2021-06-29:
You have this output? it should show more details on the error.
[image]

Reedyuk, 2021-06-29:
The iOS example needs to be built against a real device target.

Reedyuk, 2021-06-29:
Just realised now, the IOS example, you need to open the whole project in Intelli-j and build the project, this will produce a BlueFalcon framework, which you can then use. 
Here is one i made earlier.
[BlueFalcon.framework.zip](https://github.com/Reedyuk/blue-falcon/files/6732982/BlueFalcon.framework.zip)

iphonedev4radixweb, 2021-06-29:
ok let me try thanks a lot

Reedyuk, 2021-06-30:
Any lick @iphonedev4radixweb ?

Reedyuk, 2021-07-05:
Closing due to inactivity.

closed 2021-07-05 by Reedyuk