Reedyuk/blue-falcon #83: The characteristic value change delegate never fire

renatosmf, 2021-09-09:
** Describe the bug **
I'm trying to run the recording event without response and listening to the characteristic value change, but the delegate never notify the characteristic value change, but on my peripheral, I can see that it received the recording event.

**Reproduce**
Steps to reproduce the behavior:
1. Set to notify characteristic value change
2. Write about an unanswered feature
3. It should get notification of value changes, but it's not working properly.

**Expected behavior**
Receive notification about the change in the value of entered features.
 
** Smartphone (fill in the following information): **
   - Device: iPhone 12 Max Pro
   - OS: iOS 15

Reedyuk, 2021-09-10:
Can you produce a sample project with the problem?

renatosmf, 2021-09-25:
@Reedyuk You can reproduce this issue by just running your Example project trying to write the characteristic value without response and you will see that delegates method `didCharacteristcValueChanged` never is called.

nilskasseckert, 2022-04-11:
Hi @Reedyuk ,
I have a similar problem. Possibly it is related to this: I send with Response on my iPhone. I can see on the device that the message is received and would now expect the response to arrive in the "didCharacteristcValueChanged" method. But it is never called? 

Is this the same error or did I do something wrong? 

BTW: The method has a typo I just noticed

Reedyuk, 2022-04-11:
hard to tell exactly, depends on how the bluetooth device behaves.
Also depends on the device firmware and that if it broadcasts out a characteristic value changed event.

nilskasseckert, 2022-04-11:
It confirms the received data, via the standard Bluetooth way. 

But let's ask differently: how would it have to acknowledge for the library to recognize it? We develop the firmware ourselves. 

If I only use iOS and the Apple Framework, there is the possibility to send with reponse. Everything works With this way.

Reedyuk, 2022-04-11:
For when i was using the library for my own project, i was subscribing to the changes. I believe you may need to register to subscribe on value changes.
I was using this library for a simple use case, maybe yours is more complicated and requires a change the library to support such a feature.

twyatt, 2022-04-11:
> You can reproduce this issue by just running your Example project trying to write the characteristic value without response and you will see that delegates method `didCharacteristcValueChanged` never is called.

Unlike when performing a write **with response**, when performing a write **without response** the `didCharacteristcValueChanged` will not be called. This is expected behavior, [per documentation](https://developer.apple.com/documentation/corebluetooth/cbperipheraldelegate/1518823-peripheral):

> Core Bluetooth invokes this method only when your app calls the [writeValue(_:for:type:)](https://developer.apple.com/documentation/corebluetooth/cbperipheral/1518747-writevalue) method with the [CBCharacteristicWriteType.withResponse](https://developer.apple.com/documentation/corebluetooth/cbcharacteristicwritetype/withresponse) constant specified as the write type.

@Reedyuk in order to support write **without response**, Blue Falcon should expose [`canSendWriteWithoutResponse`](https://developer.apple.com/documentation/corebluetooth/cbperipheral/2891512-cansendwritewithoutresponse) and [`peripheralIsReady(toSendWriteWithoutResponse:)`](https://developer.apple.com/documentation/corebluetooth/cbperipheraldelegate/2874034-peripheralisready), then library consumers will need to wait for `peripheralIsReady(toSendWriteWithoutResponse:)` if `canSendWriteWithoutResponse` is `true` when performing a write. If `canSendWriteWithoutResponse` is `false`, then no callback will occur and writes can continue until `canSendWriteWithoutResponse` is `true`.

Reedyuk, 2022-04-11:
@twyatt - thanks for the detailed explanation. 

I'm happy for others to contribute to the repository and put in a pull request for their changes.

nilskasseckert, 2022-04-20:
@Reedyuk How to subscribe to changes? In your native code you never watch for didWriteValueFor

Reedyuk, 2022-04-20:
I believe you need to write the characteristic to subscribe.
Then the characteristic value did changed should call back on changes.

nilskasseckert, 2022-04-20:
I opened a PR for this.

closed 2022-04-20 by Reedyuk