Reedyuk/blue-falcon #85: Latest release failing due to no xcode 13 support on github

Reedyuk, 2021-12-12:
Waiting for github to support xcode 13 - blocking the release

nilskasseckert, 2022-03-28:
Hi @Reedyuk,
Anything new on that?
Regards,
Nils

Reedyuk, 2022-03-28:
Fixed, being published as we speak, should take a few hours to be able to fetch v0.10.0

nilskasseckert, 2022-04-08:
Hi @Reedyuk,

the new version can be still not found. Can you please check again?
Thanks!

Reedyuk, 2022-04-08:
https://mvnrepository.com/artifact/dev.bluefalcon/blue-falcon/0.10.0

nilskasseckert, 2022-04-08:
It seems that something else is wrong. If I use it on my KMM-Project. Version 0.9.5 is found and autocompletion works. If I now update to 0.10.0 the import is not found anymore.

Reedyuk, 2022-04-08:
does the project build?

nilskasseckert, 2022-04-08:
No because he can not found the lib

Reedyuk, 2022-04-08:
is the samples failing to build when bumping to 0.10.0?

nilskasseckert, 2022-04-08:
The sample application does not work at all anymore for me. Too old versions. Next week I can try to get it running but should be the same problem.

Reedyuk, 2022-04-08:
if i get time over the weekend, i will try and update the samples for you.

Reedyuk, 2022-04-08:
i have made a change to the lib to bump to 1.6.20 of kotlin.
I was able to get blue falcon running on 0.10.0 though.
I would recommend looking for version 0.10.1 and see if you have the same issue.
[image]

nilskasseckert, 2022-04-09:
Thanks a lot for the fast answer! It works now.

I updated the gradle version and some small things in Xcode for the iOS Project. Without the new gradle version the project does not build for me.

I openend a PR: https://github.com/Reedyuk/blue-falcon/pull/88 

Thanks a lot!

nilskasseckert, 2022-04-09:
Version 0.10.1 works also for me. Thanks

Reedyuk, 2022-04-09:
Just incase anyone else has a similar issue, blue falcon does not publish the x64 target because using intel ios simulator is pointless due to no support for bluetooth

closed 2022-04-09 by Reedyuk