Reedyuk/blue-falcon #86: BlueFalcon not found in iOS

AsifArhan, 2022-04-01:
I am getting below error while running the iOS example project
There is no XCFramework found at '/Users/apple/Downloads/blue-falcon-master/library/build/universal/BlueFalcon.xcframework'.

Reedyuk, 2022-04-01:
run the gradle task makeUniversalFramework
Then try again.

AsifArhan, 2022-04-01:
I don't have kotlin installed in my mac, do i need to download kotlin for this?

Reedyuk, 2022-04-01:
this library probably isn't for you then. It is to be used for kotlin multiplatform projects, the intention was not to be used independently on ios(technically if you had the universal framework you could make it work but probably not worth it).

closed 2022-04-01 by Reedyuk

AsifArhan, 2022-04-01:
Thank you for the quick response, is there any way to do it in iOS if you could suggest

Reedyuk, 2022-04-01:
Depends on what 'it' is, if you mean interact with bluetooth in ios? then there is plenty of great tutorials online using the sdk's provided by apple.

AsifArhan, 2022-04-04:
Okay Thank you

ZuhanLin, 2022-04-15:
> 

can not find makeUniversalFramework，pls tell me detail steps，ths

Reedyuk, 2022-04-15:
You need to run the task assemblexcframework
I need to update the readme

ZuhanLin, 2022-04-15:
> 

Thank you，i got it

ZuhanLin, 2022-04-15:
[image]
ios-arm64-simulator is needless，how to remove it？

Reedyuk, 2022-04-15:
You need to check the build gradle script and you can set the iOS target that is desired.
It's strange it didn't generate all the iOS targets. It's not much of a universal framework when it's only one target architecture

ZuhanLin, 2022-04-16:
> You need to check the build gradle script and you can set the iOS target that is desired. It's strange it didn't generate all the iOS targets. It's not much of a universal framework when it's only one target architecture

Thank you for your response，I had run successfully in my device