Reedyuk/blue-falcon #93: The project 'blue-falcon' is not a Gradle-based project

ZuhanLin, 2022-05-05:
Project Sync
The project 'blue-falcon' is not a Gradle-based project

Reedyuk, 2022-05-05:
@ZuhanLin - Any more details? the CI is building the project so i doubt its a problem on the project. Maybe an issue with your current setup?

ZuhanLin, 2022-05-05:
> @ZuhanLin - Any more details? the CI is building the project so i doubt its a problem on the project. Maybe an issue with your current setup?

i clone the project, and Sync project with gradle files by Android Studio ,

ZuhanLin, 2022-05-05:
[image]

Reedyuk, 2022-05-05:
I have just tried it with latest master and its working. 
Can you try intellij? Make sure you have the latest version, as its using Kotlin 1.6.21

ZuhanLin, 2022-05-05:
> I have just tried it with latest master and its working. Can you try intellij? Make sure you have the latest version, as its using Kotlin 1.6.21

I download the latest Intellij, but got error that 'This version of the Android Support plugin for IntelliJ IDEA (or Android Studio) cannot open this project, please retry with version 2021.1.1 or newer.
'

Reedyuk, 2022-05-05:
Run the build task, does it succeed. 
I get that error too on gradle sync, but usually the gradle build task will still succeed.

Reedyuk, 2022-05-05:
[image]

ZuhanLin, 2022-05-05:
> Run the build task, does it succeed? I get that error too on gradle sync, but usually the gradle build task will still succeed.

i can not find the gradle task list

Reedyuk, 2022-05-05:
and if you run 
```
./gradlew build
```
does it work?

ZuhanLin, 2022-05-05:
> and if you run
> 
> ```
> ./gradlew build
> ```
> 
> does it work?

Yerp, it is SUCCESS ，how can i execute special task

Reedyuk, 2022-05-05:
Special task? sounds awesome, i have no idea what you mean by special task.

maybe the question to be asked is, What exactly are you trying to do?

ZuhanLin, 2022-05-06:
i can not find gradle task list，sorry ，i am a beginner of gradle project and Intellij

Reedyuk, 2022-05-06:
When opening the root folder of the project, i would strongly recommend you open it with Intellij.
When opening the android examples, you should use android studio.

I find this works best for this project - maybe some bugs Jetbrains have not fixed.

If you open the project in intellij and press the reload button.

[image]

Then it should produce the list of tasks.

ZuhanLin, 2022-05-06:
> When opening the root folder of the project, i would strongly recommend you open it with Intellij. When opening the android examples, you should use android studio.
>
> I find this works best for this [...]

it is not list gradle task but errors below image
[image]

Reedyuk, 2022-05-06:
it could be related to the gradle wrapper you are using.
I would recommend you delete the current one and reload it to fetch a new one.

The android support plugin is very twitchy.

ZuhanLin, 2022-05-06:
How to debug with xcode or Android studio，is there any Tutorials

Reedyuk, 2022-05-06:
https://kotlinlang.org/docs/multiplatform-mobile-create-first-app.html

closed 2022-05-06 by Reedyuk