Reedyuk/blue-falcon #96: Notify not working on Android

burnhamd, 2022-07-05:
**Describe the bug**
When calling the notifyCharacteristic, the onCharacteristicChanged never firing.  I am getting a value for the read.  And I have checked that the device indeed does support notifications using the LightBlue Application on IOS.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to the device using blueFalcon.connect(bluetoothPeripheral, true)
2. Wait until didDiscoverServices Returns
3. Search through service to find the characteristic I want to nofify4. 
4. Call notifyCharacteristic on the returned characteristic 

`      char = service.characteristics.first { it.name == TESS5600_DATA_CHAR.lowercase() }
        bluetoothService.characteristicDelegate = this
        bluetoothService.notifyCharacteristic(bluetoothPeripheral, char!!, true)`

**Expected behavior**
Expecting the onCharacteristicChanged to fire

**Smartphone**
 - Device: Samsung Tablet
 - OS: Android API 31

Reedyuk, 2022-07-05:
Can you try the same thing using lightblue on Android(as you did for ios)? this rules out any reason it could be an android thing.

I would also suggest, after you request 'notify characteristic', can you then perform a single read after this, then wait to see if it updates?

burnhamd, 2022-07-05:
For some reason light blue is crashing on this device on startup.  Will see if I can fix that and test.

If I do a read after notify, then I get no read result either.

burnhamd, 2022-07-05:
Just confirmed that when I subscribe using the light blue app the notifications are received from the peripheral.

burnhamd, 2022-07-05:
I think the issue is here:
`it.characteristic.descriptors.forEach { descriptor ->
                        descriptor.value = descriptorValue
                        gatt.writeDescriptor(descriptor)
                    }`

Here it is iterating through all descriptors and attempting to write an enable notification to it.  But I have additional descriptor.  Instead I think it should just set the notify descriptor "00002902-0000-1000-8000-00805f9b34fb"

leonardovallem, 2024-05-13:
Hello you all! Any updates on this?

I'm having the exact same issue. Works on iOS, but on Android it doesn't. It does works sometimes, but most of times it doesn't. onCharacteristicChanged is not being called.

Any ideas?

Thanks in advance.

Reedyuk, 2024-05-16:
I havent had chance to look into it. I also don't have the exact devices and setup to reproduce so it would be difficult for me. I am happy for anyone to produce a solution in a PR and i can review it.

Reedyuk, 2025-02-03:
Android is quite picky with regards to notify.

From past experiences, i have found that you should make sure you make a characteristic as the right type, for example using the notify and indicate, on a just notify characteristic, generally means it will fail and not send any notifications.

Notiftandindicate is not a golden ticket when it comes to android and the device it is connecting to, so make sure you do this correctly.
Secondly, be careful with firing off requests/changes in quick succession, i have found that android can drop requests, if you spam the device with too many requests.

closed 2025-02-09 by Reedyuk