Reedyuk/blue-falcon #97: Second disconnect not working (Android)

MadMe86, 2022-07-20:
**Describe the bug**
A second disconnect from a bluetoothPeripheral is not working. Version 0.10.7
Maybe related to #53 ?

**To Reproduce**
Steps to reproduce the behavior:
1. Scan for peripherals
2. connect
3. (read some characteristics)
4. disconnect
5. connect again
6. try to disconnect. 

The device is afterwards still connected.
I could also reproduce this behavior with the Android example app.

**Expected behavior**
The device is disconnected after disconnect is called.

**Smartphone (please complete the following information):**
 - Device: Samsung A52
 - OS: Android 12

I checked the code and found maybe the reason for this behavior:
When disconnect is called the disconnect and close methods of the gatt object got called. But it got not removed from gatts.
If the same device is connected again it retrieves for disconnecting the same gatt object. Now we have two GATT objects for the same bluetooth device.
I made a fix which works for me. 
In short: removing the GATT objects in the disconnect method after disconnect() and closing(), also closing the corresponding GATT after state change to STATE_DISCONNECTED.
PR is incoming :)

MadMe86, 2022-07-21:
Thanks for mergeing :)
0.10.8 is at the moment not available on Maven Central... does the release take a while?

Reedyuk, 2022-07-21:
Sorry, i forgot to release on sonatype. Doing this now, so you should be able to pull it in around an hour, it will appear on maven central a few hours later.

closed 2022-07-21 by Reedyuk