Reedyuk/blue-falcon #100: Add information about Bluetooth Classic to the README or change terminology to "BLE"

okarmazin, 2023-03-25:
The README is confusing. It insists on using the word "Bluetooth" (which means Bluetooth Classic), but then all examples are given in terms of BLE. They are not interchangeable.

Does this library implement Bluetooth Classic, or only BLE? The README doesn't say.

Reedyuk, 2023-03-25:
You should be able to discover classic bluetooth devices and connect to them.
What exactly are you having trouble understanding? 
The library is really just a wrapper of the pre-existing bluetooth services offered by the platforms. If there is something missing, it should be easy enough to add.

Reedyuk, 2023-03-25:
actually i stand corrected, you are right.
Currently the library doesnt support classic bluetooth, only BLE.
Happy for you to write the code to support classic bluetooth though.

closed 2024-08-08 by Reedyuk