Reedyuk/blue-falcon #106: Kotlin MP Example

nicolasf, 2023-10-18:
The Kotlin MP example apparently was deleted from the repository, but it's mentioned on the documentation.
Was this intentional?

Reedyuk, 2023-10-18:
whoops, let me put it back

Reedyuk, 2023-10-18:
Actually i don't think we ever had a KMM example. I need to do one for you.

nicolasf, 2023-10-18:
Maybe it was deleted at this commit? https://github.com/Reedyuk/blue-falcon/commit/9bf7e4b80f497ac541a8e1605969209285f2a494

Aleksandar98, 2023-10-26:
Hey @Reedyuk any news about KMM example?

Reedyuk, 2023-10-26:
Sorry, have been busy recently.
It's pretty easy though, include package in common and then instantiate blue falcon in the common.

Reedyuk, 2023-10-28:
Hi guys, i have added a very basic example now, a couple of notes:
1. Run on a real device - emulator/simulator wont work
2. When you press scan, it will output the results to the log
This should atleast point you in the right direction.
Let me know if this works for you guys

Aleksandar98, 2023-10-29:
Thanks @Reedyuk , will be trying it out this week

Reedyuk, 2023-10-29:
Great, good luck and enjoy. Let me know if anything needs changing. 
I will close this issue and you can raise an issue or discussion if you need anything

closed 2023-10-29 by Reedyuk