Reedyuk/blue-falcon #113: Allow log configuration

SarahDelCastillo, 2023-12-14:
When scanning with BlueFalcon, the console is completely flooded with logs like this:

```Discovered device null
Discovered device null
Discovered device DatiPlus-K00001B
Discovered device DatiPlus-K00001B
Discovered device null
Discovered device null
Discovered device Apple TV
Discovered device Apple TV
Discovered device null
Discovered device null
Discovered device null
```

It would be great if we could configure/deactivate logging.

I tried just filtering on the console but its not a viable solution if I need to see other similar logs.

Reedyuk, 2023-12-14:
Good point, we should probably expose some sort of logger interface and then allow the consumer of the library to set/handle the logs

Reedyuk, 2024-08-08:
@SarahDelCastillo - this is now complete v 1.2.0

closed 2024-08-08 by Reedyuk