Reedyuk/blue-falcon #115: Addition of Compose Multiplatform example project

neelratanguria, 2024-02-16:
It would be really helpful if this repository provided an example of Compose Multiplatfrom project as well. FYI, I managed to get the CMP project running fine, but it costed me more than 4hrs. I would love to contribute the boilerplate this repo.

Quick note: I was really confused when initializing the BlueFalcon object on iOS side, because the way it was inject directly into the ViewModel in KMP project. Then, I realised that ContextApplication can be garbage for initializing the iOS implementation. Here is how it looks like

AppModule.ios.kt
```
actual class AppModule {
    actual val blueFalcon: BlueFalcon
        get() = BlueFalcon(ApplicationContext(), null)
}
```

AppModule.android.kt
```
actual class AppModule(
    private val context: Context
) {
    actual val blueFalcon: BlueFalcon
        get() = BlueFalcon(context as ApplicationContext, null)
}
```

Reedyuk, 2024-02-16:
Hi @neelratanguria - i dont have the time to do a compose multiplatform example but welcome any pr's that introduce one.
As for the context, its probably not very clear in the docs. You can pass null, nsbundle or even UIApplication.

neelratanguria, 2024-02-17:
I will do it then i guess

neelratanguria, 2024-02-18:
Hi @Reedyuk,
I have created a [pull request](https://github.com/Reedyuk/blue-falcon/pull/116) for the discussed example. Please do review it.

Reedyuk, 2024-02-18:
Many thanks for your change, i have merged in.

closed 2024-02-18 by Reedyuk