Reedyuk/blue-falcon #117: GPLv3 is a pain

9il, 2024-04-13:
Hi,

We have started to work on the Blue-Falcon extension, and then I found that it is GPLv3.
It is an absolutely restrictive and unclear license for commercial applications.

I am happy to share all the changes we will make to Blue-Falcon.

However, GPLv3 seems incompatible with Apple's policies. Plus, we definitely can't open-source our mobile application. We can publish only changes we will make for Blue-Falcon.

Is GPLv3 mandatory?

Kind regards,
Ilia

Reedyuk, 2024-04-13:
I can change the licence, it's not a problem, what licence would be best for you

9il, 2024-04-13:
Thank you, that will be very helpful. We will add L2CAP API Blue-Falcon.

I use [Apache License, Version 2.0](https://www.apache.org/licenses/LICENSE-2.0) for my open-source projects, such as [mir-algorithm](https://github.com/libmir/mir-algorithm). It would be awesome to have this license here too.

Reedyuk, 2024-04-13:
is this ok?
https://github.com/Reedyuk/blue-falcon/commit/f8cbb62bffaa9913f25171aa2e19cca612c4af73
Let me know and if so close this issue.

9il, 2024-04-13:
The commit adds a license text, but there is no copyright.

Instead of adding the text, Apache suggests adding the following (edit year and owner):

```
Copyright [yyyy] [name of copyright owner]

Licensed under the Apache License, Version 2.0 (the "License");
you may not use this file except in compliance with the License.
You may obtain a copy of the License at

    http://www.apache.org/licenses/LICENSE-2.0

Unless required by applicable law or agreed to in writing, software
distributed under the License is distributed on an "AS IS" BASIS,
WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
See the License for the specific language governing permissions and
limitations under the License.
```

9il, 2024-04-13:
The suggestion can be found here https://www.apache.org/licenses/LICENSE-2.0#apply

Reedyuk, 2024-04-13:
ah my bad, done

closed 2024-04-13 by 9il