Reedyuk/blue-falcon #119: Bonding callback

kcn1-71, 2024-04-29:
Hello guyz! Thank you for this library! 
It there any way to check if bonding is finished?

Reedyuk, 2024-04-29:
There is no official way, someone suggested this for Android

https://stackoverflow.com/questions/33317516/android-ble-how-to-check-if-bonded-ble-device-is-available-to-connect

But I wouldn't want to include this in the library as it's not official and is a bit of a hack. Plus could break with any update of Android.

Then the headache to support it also on iOS too.

kcn1-71, 2024-04-29:
I found out that in onConnected callback, on Android side, peripheral.bluetoothDevice.bondState is available, but I didn't find any ways to implement the same on iOS side :(

Reedyuk, 2024-04-29:
Unfortunately i'm not sure there is the equivalent for ios, we would need the same functionality on both platforms, otherwise it shouldn't be implemented.

kcn1-71, 2024-04-29:
Got it... I'll try to make it on server side... maybe some not encrypted characteristic with current authorisation status

closed 2024-04-29 by kcn1-71