Reedyuk/blue-falcon #120: Bluetooth not work on Android 11 API 30

youvita, 2024-04-30:
Any help, I when clicked scan is started status. However don't display any bluetooth found.
Noted: Bluetooth and permission is turn on status

[image]
[image]

Reedyuk, 2024-04-30:
It would appear that bluetoothadapter.java is throwing an exception. If you click into it you will see the code and find why this is the case. 
Usually the stacktrace is bigger and you will find the original cause lower down(its cut off on your screenshot).
Chances are, this is an issue with android.bluetooth.BluetoothAdapter(standard android) and not an issue with the blue falcon library.
It could be a permissions issue, seems like there is a fetching of systemService from a context wrapper. - Permission with xml manifest?

youvita, 2024-04-30:
Yes, I also added the permission in manifest. However I tried with other device android version 14 API 34 it's working. That's why I think the issue come from version android device.

Reedyuk, 2024-04-30:
So it works on version 14 API 34 of android. Which version of android does it not work? What is the full stack trace? you will probably find a hint of the error there

youvita, 2024-05-02:
The currently android version 11 API 30 not work. The actually I try run the same source code but the version 11 of android not list bluetooth found. however it's work on version 14 API 34
Noted: both device also turn on bluetooth and clicked scan button.

Device not work
[image]

Device work
[image]

Reedyuk, 2024-05-02:
Do you have the **full** stacktrace? could you post it in here?

youvita, 2024-05-02:
I'm not sure what is the full stack trace. However I have to explore logs file, may it's can help.

[OnePlus-ONEPLUS-A6000-Android-11_2024-05-02_152223.logcat.zip](https://github.com/Reedyuk/blue-falcon/files/15185633/OnePlus-ONEPLUS-A6000-Android-11_2024-05-02_152223.logcat.zip)

Reedyuk, 2024-05-02:
Having a look now for you

Reedyuk, 2024-05-02:
Are you sure you have the correct permissions:
```

    /**
     * Broadcast Action: The notifys Bluetooth ACL connected event. This will be
     * by BLE Always on enabled application to know the ACL_CONNECTED event
     * when Bluetooth state in STATE_BLE_ON. This denotes GATT connection
     * as Bluetooth LE is the only feature available in STATE_BLE_ON
     *
     * This is counterpart of {@link BluetoothDevice#ACTION_ACL_CONNECTED} which
     * works in Bluetooth state STATE_ON
     *
     * @hide
     */
    @RequiresBluetoothConnectPermission
    @RequiresPermission(android.Manifest.permission.BLUETOOTH_CONNECT)
    @SdkConstant(SdkConstantType.BROADCAST_INTENT_ACTION)
    public static final String ACTION_BLE_ACL_CONNECTED =
            "android.bluetooth.adapter.action.BLE_ACL_CONNECTED";
```

My guess is that you dont have the bluetooth_connect permission.

`public static final String BLUETOOTH_CONNECT = "android.permission.BLUETOOTH_CONNECT";`

youvita, 2024-05-02:
Yes, I'm sure. Because I'm just from from your source code.

[image]

Reedyuk, 2024-05-02:
i would need to spend time looking into this properly, will have to try later when i have a moment. My guess is its something around contexts and permissions used in an older api of android.
Not the best solution but i would propose you don't support the older api bump the minimum up.

youvita, 2024-05-02:
Thanks for your help

Reedyuk, 2024-05-02:
@youvita - i have tried to recreate the issue but because i dont have a physical android device on android 11, i cannot reproduce. The emulator doesnt crash but obviously it doesnt engage with bluetooth.
As a suggestion i could bump the min api level so that android 11 cannot be supported but that wont fix your problem of supporting those devices.

youvita, 2024-05-13:
@Reedyuk,
I have question, it's possible to display bluetooth name instead bluetooth address?
Example: 77:D9:EE:E6:E2:52 -> U6-LR

Reedyuk, 2024-05-13:
Ble devices expose a characteristic called name, you must call this and then in your app, display the value of this characteristic.

youvita, 2024-05-16:
@Reedyuk, Could you update this name bluetooth address instead characteristic as you mention above? 
It's better know the name of Bluetooth device. Thank you for feedback.

[image]

Reedyuk, 2024-05-16:
No, as this is only an android thing and also this could have an impact on other users of the library.
It is up to you, the consumer of the library to do this your side.

youvita, 2024-05-17:
It's possible, you can mention the example code here. Thanks for support.

Reedyuk, 2024-05-17:
Sorry, you will have to do that yourself.

The sudo is rather simple:
Detect device, request characteristic of 'name', when you get the value then display.

You could even extend the BluetoothPeripherial, to have a suspend func to get the Mac address.

Then in your client code, you could have a flow of devices mapped to by key to device e.g.
```
val deviceMap: MutableStateFlow<Map<String, BluetoothPeripherial>> = blueFalconClient.peripherals.map {
it.macAddress() to it
}
```

closed 2024-08-13 by Reedyuk