Reedyuk/blue-falcon #123: iOS / Android diferncies

kcn1-71, 2024-05-15:
Hey guys! Maybe you can help me

I'm using your library in a multi-platform application, all logic is shared

There is a problem that the library does not send requests to the connected device.

On Android, I can scan a device, connect to it, request a characteristic reading, see the read request logs on the gatt server, get the reading result in didCharacteristcValueChanged

On iOS I can scan the device, connect to the device, enter the bonding code, after that I try to request a reading of the characteristics

From the logs I see that
client.readCharacteristic(peripheral, сharacteristic) is called, but the call does not reach the client (I have a Gatt server on ESP32) and of course didCharacteristcValueChanged is never called

Tell me, maybe there is some trick here for iOS?

Reedyuk, 2024-05-15:
can you confirm.
In your common code you are connecting and sending the request to read the characteristic, this is working in android but not in ios?

I would recommend you clone this repo, publish to your maven local, you could then try and add additional logging to locate where things are going wrong.

Here is a question, can you read ANY characteristic? I think because its a characteristic that needs to be 'subscribed' to, (on iOS), you need to perform "notifyAndIndicateCharacteristic".
I had a similiar issue where i could read some characteristics but others i couldn't and it was because they need to be set to notify of a change - a behaviour that was slightly different on android.

kcn1-71, 2024-05-15:
> can you confirm. In your common code you are connecting and sending the request to read the characteristic, this is working in android but not in ios?
> 
Correct

> Here is a question, can you read ANY characteristic? I think because its a characteristic that needs to be 'subscribed' to, (on iOS), you need to perform "notifyAndIndicateCharacteristic".
> I had a [...]

I tried to read encripted and not encripted characteristic, tried to client.notifyAndIndicateCharacteristic(connected, characteristic, true) before request, but it doesn'work

> I would recommend you clone this repo, publish to your maven local, you could then try and add additional logging to locate where things are going wrong.

Can you recomend some specific place frome where is better to start checking?

kcn1-71, 2024-05-15:
I did publishToMavenLocal and got this set of folders 
[image]

I don’t understand which one I need to connect to my project? Or do I need to connect each of them?

Reedyuk, 2024-05-15:
You do not need to worry about which to pick, gradle will deal with that.
You just need to add
```
mavenLocal()
```
in your build gradle for your project, i used it in the android example:
https://github.com/Reedyuk/blue-falcon/blob/458659a05ab07133dc7c4ad10d567d2b076c0ab3/examples/Android-Example/build.gradle#L25

kcn1-71, 2024-05-15:
Ok, so I added some logs in iOS side of library

_connect
disconnect
scan
stopScanning
changeMTU_
Called as expected

_readCharacteristic
notifyCharacteristic
indicateCharacteristic
notifyAndIndicateCharacteristic_
Never called

Reedyuk, 2024-05-15:
even the basic readCharacteristic is not being called? Are you sure you are calling it correctly? I would imagine other users of the library would of raised this issue if this was the case.

kcn1-71, 2024-05-15:
Ok, I have localized the error

In my gatt server, all names of services and characteristics are hardcoded with lines like
620b871f-0002-4013-9c5b-ff5b98eedacb

Callbacks on the Android side return the names of services and characteristics in the form
620b871f-0002-4013-9c5b-ff5b98eedacb

And callbacks on the iOS side return them in the form
620B871F-0001-4013-9C5B-FF5B98EEDACB

The search was carried out by comparing strings, I simply copied all the names from the server to my client, and on Android they matched, but not on iOS

kcn1-71, 2024-05-15:
This may all seem stupid, but it was quite unexpected

Reedyuk, 2024-05-15:
oh what a pain, i think i have had a similar issue in the past, i would be tempted to force upper case but wasn't sure it was a good idea.

Reedyuk, 2024-05-15:
Can you close the issue if you feel the problem is fixed

closed 2024-05-15 by kcn1-71

BenMorrisRains, 2025-01-12:
I think this issue should be resolved/fixed in the lib. This bit me for a few hours the other day.

Reedyuk, 2025-01-12:
Unfortunately ben, we cannot do this, its not just a case of upper case and lower case.
Ios adopts using a different shorthand notation, android uses the full 32bit uuid.
The best we can do, which i have changes in the latest is to have an expect and actual type of uuid, where ios is cbuuid and android is using kotlin uuid.