Reedyuk/blue-falcon #124: Connection disables when app is kept in background

vishwaprojects, 2024-05-21:
**Describe the bug**
The connection is succesfull, but the moment ,I keep the app in background the connection is refused and disconnected automatically (trying with esp32) 

**To Reproduce**
Steps to reproduce the behavior:
1. connect to the ble 
2. keep the app in background(press home button)

**Expected behavior**
perssist connection even with the app is in background

**Smartphone (please complete the following information):**
 - Poco c3
- Android version 12

**Additional context**
Add any other context about the problem here.

Reedyuk, 2024-05-21:
What makes you think that this wrapper library is performing a disconnect? seems more likely the implementation is not keeping the connection open.
Do you have any logging in place on the device to see if the device is being disconnected?

vishwaprojects, 2024-05-22:
yeah forgot to mention below is the log i get when i press button 
W (711892) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x13 W (711892) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x13
I am completely new to kmm and ble

Reedyuk, 2024-08-08:
Ok, i think i know your issue,
https://github.com/Reedyuk/blue-falcon/blob/1f6960d4bab9e98f792354a11cbcf431c7f7429a/library/src/commonMain/kotlin/dev/bluefalcon/BlueFalcon.kt#L19

You need to make sure that you pass TRUE for autoConnect.

```
blueFalcon.connect(device, true)
```

I believe you are probably using the default value which is false.

closed 2024-08-08 by Reedyuk