Reedyuk/blue-falcon #126: Make Android BluetoothPeripheral Serializable/Parcelable

leonardovallem, 2024-06-10:
**Is your feature request related to a problem? Please describe.**
I'm using BlueFalcon in a Compose Multiplatform project along with Voyager for navigation. Since Voyager requires their screens to be Serializable, having an argument of type `BluetoothPeripheral` makes the application crash on Android when it goes to background.

**Describe the solution you'd like**
Making the actual class `BluetoothPeripheral` for androidMain implement `java.io.Serializable`.

**Describe alternatives you've considered**
Making the actual class `BluetoothPeripheral` for androidMain implement `Parcelable`. Could be easy since its property **bluetoothDevice** is of type `android.bluetooth.BluetoothDevice`, which is Parcelable.

**Additional context**
-

Reedyuk, 2024-06-11:
I like the idea but, I am weary of adding another dependency for other consumers of the library, the intention is to be a lightweight wrapper.

I guess you can't map your own parcelable object to this Bluetooth peripheral. 
Maybe I could make the Android implementation parcelable.

leonardovallem, 2024-06-11:
Yeah... I could be wrong but I think it wouldn't cause any troubles.

Reedyuk, 2024-06-11:
Any good @leonardovallem ?
https://github.com/Reedyuk/blue-falcon/pull/127

leonardovallem, 2024-06-11:
Amazing! Thank you so much!

Reedyuk, 2024-06-12:
can you try out v 1.0.1 and re-open this issue if it doesnt work

closed 2024-06-12 by Reedyuk

leonardovallem, 2024-06-12:
working perfectly fine. thank you so much for the effort!