Reedyuk/blue-falcon #132: KMP CBCharacteristic only read value

TimurGaysin, 2024-07-10:
In Swift version,  CBCharacteristic have read/write type. (https://developer.apple.com/documentation/corebluetooth/cbcharacteristic)https://developer.apple.com/documentation/corebluetooth/cbcharacteristic/value
In blue-falcon KMP version, CBCharacteristic  - have only read type(it defined as val) - CBCharacteristic.value cannot be reassigned.

Reedyuk, 2024-07-10:
It is a computed property, Computed properties in Kotlin do not store a value directly but compute it on-the-fly using a getter. 

https://github.com/Reedyuk/blue-falcon/blob/2448f8f1ed17393ad88a536d50f14e404a6f59ff/library/src/appleMain/kotlin/dev/bluefalcon/BluetoothCharacteristic.kt#L15C5-L22C10

Reedyuk, 2024-07-10:
can i ask, where are you getting "value cannot be reassigned"? are you saying this is ocurring during runtime?

Reedyuk, 2024-08-08:
closing during inactivty

closed 2024-08-08 by Reedyuk