Reedyuk/blue-falcon #133: Add "library" source example

Tabaqui, 2024-07-19:
Hello!

Please provide a solution of how to add "library" sources to basic kmp project and use it instead of a prebuild dependency.

I'm trying to fix a few of problems.

While trying a little bit customed "KotlinMP-example" I get spamming from blueFalcon.perepherals.collect { .. }. collect method is called indefinite number of times which I guess is ok and it gives me a set of devices. This set is increased every call by one with as I assume same device somehow copied for unknown for me reason. So spending time I get sets containing huge number of devises like thousands. I suspect it is somehow related to my device(Xiaomi 12 pro). I want to find out how.
So now I find a way to debug and fix the issue at least locally.

Another thing where it might be helpful.

I have another device. I have also tested with it but I have had no luck(no any calls, even permission requests). I conclude it happened because the device is too old. It is Xiaomi MI 8.
I see some limitations in the code. It is minSdk=24 but a firmware is based on Android 10 not 7.
Here I also would like to understand what's what.

Please point out a direction. Thanks!

Reedyuk, 2024-07-19:
ok a few questions to deconstruct here.

1. Solution to add library source. - I need to get some spare time to make the example cleaner and tidy things up, this project has pivoted from being used directly in native, to being used ONLY in KMP. The intention is that you should be able to include blue-falcon in your KMP project, to produce **one** simple api for all bluetooth actions.
2. There very well could be a bug with the flow, it was something i put in to make the project more 'reactive' but i should of marked it as unstable, as i havent tested it thoroughly.
3. How to debug? - Maybe i should write a section in the readme on this, the way i have debugged in the past, checkout the blue-falcon project, publish to maven local, then make sure your KMP project has MavenLocal() in the repositories of the root build gradle/settings. Then when you put a debug point in, it should jump into the source code, you can then put in debugging. I also was thinking of rolling out a logger, which you could plug into, as its quite noisy otherwise.
4. Android 10 support - Yes this should be working, it must be a permission issue somewhere, this is usally what happens, again if you do the debug, you will be able to break point into where an exception happens upon performing a scan, there could be a custom exception that is being swallowed and ignored.

I hope that gives you some insight, its a tricky thing with bluetooth because you cant easily emulate issues, you usually have to have the device at hand, which makes it more difficult for myself to try and investigate and resolve the issues

Tabaqui, 2024-07-19:
> I hope that gives you some insight, its a tricky thing with bluetooth because you cant easily emulate issues, you usually 
> have to have the device at hand, which makes it more difficult for myself to try and investigate and resolve the issues
I see that cumming. Hope I will have good news soon.

Thanks for so fast reply and the insights. Logging would be helpful.

Tabaqui, 2024-07-19:
Since then.

I was able to find all the reasons. There are all pretty small. There all related to Android only. 

What should I do? PR?  
There all pretty small but not all of them are easy to fix. It is sad

Reedyuk, 2024-07-19:
Pr would be great, thanks

gitbiagio, 2024-08-02:
> While trying a little bit customed "KotlinMP-example" I get spamming from blueFalcon.perepherals.collect { .. }. collect method is called indefinite number of times which I guess is ok and it gives [...]

I ran into the same issue and found a solution by filtering out duplicates from the peripheral set. However I don't know if it was intended by design or if it is really an issue... I just created a PR with my modifications, hope they are useful to someone.

Tabaqui, 2024-08-04:
@gitbiagio Hi
Unfortunately due to different issues I haven't been able to get back to it and I won't some time more. 
I couldn't find the PR so I've decided to put my question over here.

Was that something related to hash of BluetoothPeripheral there? I found that it troubles when it tries to add new object to the set which supposed to be the same but counts as different

gitbiagio, 2024-08-05:
@Tabaqui Hi, you're correct: because BluetoothPeripheral is not a data class, it has no hash function and every instance is different from each other. Building an hash function doesn't resolve the case however, because it ends up adding only the first advertisement seen and never refreshing other data coming with new advertisements (e.g. the rssi value). What I've done is to manually update the set by removing old entries with same uuid before adding new ones.

Look at the PR here #134

Reedyuk, 2024-08-07:
Hi guys, 
So i have pushed a new change, where i have overridden the equals and hashcode of the bluetooth peripherial, this should stop the excessive reloading when the device has not changed.

I managed to come across an issue with the ios example, because i was using .name as the ID in swiftui, it was rendering the same device multiple times, instead when i use the UUID as the id, it works correctly.

Reedyuk, 2024-08-08:
Can i ask, what exactly do you mean by this?
"Please provide a solution of how to add "library" sources to basic kmp project and use it instead of a prebuild dependency."
As i want to help but not sure on the exact question.
Do you want to know how to use KMP libraries in your own KMP library?
Do you want to know how to use KMP libraries in your own KMP project?
Do you want to know how to use KMP libraries in your own native project?

Tabaqui, 2024-08-09:
I'm not sure you were saying about me.
If so the case you explained was fine enough. Despite that I have had  some experience in Android I forgot how to act with libraries at all. Also as I said I have been facing some troubles with physical abilities  lately which makes me so slow at some moments.
Well, thanks for helping indeed.

Reedyuk, 2024-08-09:
Based on what you have said, it sounds like in general you are struggling with including libraries in your project and this issue is not related to blue falcon but a fundamental in software development.
I will be closing this issue.

closed 2024-08-09 by Reedyuk