Reedyuk/blue-falcon #135: Need help in compose

kameshkarthik, 2024-08-08:
Hi, 
I am new to compose and BLE. Can you kindly help me how to achieve the following:

Requesting data using GET operation:
;;#REQSTART<any integer 0-64>
GET <SPACE> <CODE>àCodes can be referred in the below parameters table.
;;#REQEND <same integer should be used which is used on top for REQSTART >  

Similarly Set
;;#REQSTART< any integer 0-64>
SET <SPACE> <CODE> <SPACE> <MODIFIED VALUE>àCodes can be referred in the below parameters table.
;;#REQEND <same integer should be used which is used on top for REQSTART >  

I have the CODE values.

Any help would be appreciated .

Reedyuk, 2024-08-08:
What do you mean by get operation?
BLE devices(**Peripherials**) have **services** and those services have **characteristics**

Based on what you have said, it sounds to me like you want to do the following:
Discover a peripherial
Connect to a peripherial
Find the characteristic you want to get a value from(do this by checking all services characteristics)
get value for characteristic.

As for compose, that is just simply for UI, please look online for many courses on how to use Compose. For references, look at the Example here: 
 https://github.com/Reedyuk/blue-falcon/tree/master/examples/ComposeMultiplatform-Example

I hope this helps

closed 2024-08-08 by Reedyuk

kameshkarthik, 2024-08-08:
Sorry, couldn't put it together. 
How can I achieve reading values when in know serviceUUID and characteristicUUID? 
read(deviceId: string, serviceUUID: string, characteristicUUID: string): Promise<any>;

Reedyuk, 2024-08-08:
When you detect a device(Bluetooth Peripherial),
do a println on your  
```
device.services 
```
This will output all the services on your BLE device, you will then be able to perform the same on the characteristics.

What BLE device are connecting to? you must have documentation from the manufacturer, they will provide a FIX service and characteristic UUID.