Reedyuk/blue-falcon #148: Detailed documentation / Completed examples

aleksastanic, 2024-11-22:
Is there going to be a more detailed documentation (ex. step-by-step for all functionalities) any time soon?

Also, I don't think compose multiplatform example is completed, it does not demonstrate all possiblities of this library. Is there a plan to finish it soon?

Thank you for your work!

Reedyuk, 2024-11-22:
Hey, 
Thanks for the feedback. The functionality is there, i guess documentation could be updated.
What more do you want the example to do?

aleksastanic, 2024-11-25:
I did not find description for any of BlueFalcon methods, how and for what they are used. I did manage to writeCharacteristic (change it's value) and readCharacteristic (read it's current value), but I don't know how to use notify or indicate characteristic. When I trigger them - descriptor is changed, but I don't know how can i subscribe to receive the characteristic value changes?

Reedyuk, 2024-11-25:
The readme has some description and the website is autogenerated from it here: https://bluefalcon.dev

It should behave similar to that of the android sdk. Granted the documentation is not perfect, hence we have examples to aid understanding.

BenMorrisRains, 2025-01-12:
Hi Reedy, I appreciate your work here. Myself and a coworker have been attempting to use the library for last few months and have made some decent headway. I would agree though that some better examples or tutorials would be great for getting people up to speed faster. 

I'm not a BLE expert by any means and this library has been easier to use than others I looked at, so I think this library is on the right track. Once we are done if I am comfortable enough I wouldn't mind contributing some examples or making some YouTube videos to try to get people going faster. We have had to do a lot of self-discovery which has been informative but has taken a decent amount of time.

Reedyuk, 2025-01-12:
Yes please, any contributions are welcomed.