Reedyuk/blue-falcon #153: Data Conversion Issue When Sending Hexadecimal Data via sharedWriteCharacteristic Method

ljiezhou, 2024-12-29:
I am encountering an issue while using the sharedWriteCharacteristic method to send data. The raw data I want to send is a hexadecimal array, such as:

A5 45 19 00 F0 1E 05 88 13 00 00 64 00 00 00 05 A0 58 39 00 00 00 00 00 00 00 00 DC 1E 5B

However, after converting the data to a ByteArray and calling the sharedWriteCharacteristic method, the data is automatically converted into a string, causing the actual sent data to not match my requirements. The data that gets sent becomes:

7055555a41504165425967544141426b41414141426142594f514141414141414141414133423562

The issue is in this line of code:
```
sharedWriteCharacteristic(bluetoothPeripheral, bluetoothCharacteristic, value.toString(Charset.defaultCharset()), writeType)
```

It converts the data I want to send into a string, which changes the data format. How can I ensure that the hexadecimal data is sent as-is, without being converted into a string? How can I avoid the data changes caused by character encoding during this process?

I look forward to your help. Thank you!

Reedyuk, 2024-12-29:
Please call

writeCharacteristicWithoutEncoding

This should ensure it doesnt happen

ljiezhou, 2024-12-29:
Sorry, I didn’t check the API carefully. It turns out that writeCharacteristicWithoutEncoding can send the data correctly.

Reedyuk, 2024-12-29:
Let me know if it works and close the issue if so

ljiezhou, 2024-12-29:
It works and sends data as expected. Thank you!

closed 2024-12-29 by ljiezhou