Reedyuk/blue-falcon #154: MTU Configuration and Data Transmission Challenges on Android and iOS

ljiezhou, 2024-12-29:
Currently, I have verified that setting the MTU on Android devices does not seem to take effect as expected. For iOS, further validation is required.

To send data successfully, I need to limit the size of the ByteArray to within 20 bytes. If the data is too large, it must be split into smaller packets, and an interval must be set between packets; otherwise, transmission fails.

Additionally, there is packet loss when receiving data. My Bluetooth device is configured to transmit data to the phone with a packet size of 247 bytes.

On Android, I configure the MTU using the following method, which works stably without any packet loss:
```
override fun onServicesDiscovered(gatt: BluetoothGatt, status: Int) {
    LogUtils.d("onServicesDiscovered", status)
    if (status == BluetoothGatt.GATT_SUCCESS) {
        gatt.requestMtu(MTU)
    }
}
```
In this callback, the MTU is set to 247, as my Bluetooth device supports a maximum MTU of 247.

Based on my research:
	•	Android can set the MTU up to 512 bytes.
	•	iOS devices, however, can only set the MTU up to a maximum of 185 bytes.

ljiezhou, 2024-12-30:
Using the same Android phone and the same Bluetooth device, I conducted tests by sending the same command:
A5 45 19 00 F0 1E 05 D0 07 00 00 64 00 00 00 0A A0 53 39 00 00 14 00 14 00 F4 01 DC 1E B4.

Observations with Android Native Code:
	•	Sending a ByteArray larger than 20 bytes works perfectly fine.
	•	For receiving data exceeding 1MB, each received packet size is 247 bytes, and all data is received successfully.

Observations with BlueFalcon:
	•	When sending data, a ByteArray larger than 20 bytes requires splitting into smaller packets. However, if the ByteArray size is around 10 bytes, it works without issues.
	•	For receiving data, due to the large amount of data, packet loss occurs consistently.

Testing Conditions:
	•	The MTU was set in both cases.
	•	Testing was performed under identical conditions.

With Android native code, everything works correctly, including both sending and receiving data. However, using BlueFalcon, issues arise when handling larger ByteArray sizes and when receiving large volumes of data.

Reedyuk, 2025-01-02:
ok looks like some investigation is required, as it should be able to do this.

Reedyuk, 2025-01-02:
Could be related to the same issue here: https://github.com/Reedyuk/blue-falcon/issues/63

ljiezhou, 2025-02-09:
Dear Author,

First of all, I would like to express my heartfelt thanks. I previously reported an issue regarding data sending and receiving on the open-source platform, and I am happy to see that the update you released has effectively resolved the issue. After testing, I have noticed a significant improvement in the stability of data transmission and reception.

I will continue with further testing and stay updated on your project. If there are any new issues or feedback, I will make sure to reach out to you in a timely manner.

Once again, thank you for your hard work and contribution!

Best regards,

ljiezhou, 2025-02-09:
I’m glad to see your new update, and I tested the latest version. The command sending is working fine now. However, when receiving data, the packets are fixed at 20 bytes each. I still need to confirm the reason for this, because the same device works correctly with the native Android code.

Reedyuk, 2025-02-09:
I think the limitation of 20 bytes being sent is coming from the connected device.
You can only infer the mtu size to a device, but it is entirely up to the device if it wishes to respect it.
I am closing this issue, raise another if you suspect it is to do with the blue falcon code

closed 2025-02-09 by Reedyuk