Reedyuk/blue-falcon #156: Unable to Discover Services and Characteristics on Arduino BLE Device

pesto13, 2025-01-08:
### **Title:** Unable to Discover Services and Characteristics on Arduino BLE Device

---

**Hello,**  

I am using the BlueFalcon library for Bluetooth Low Energy communication, and I am encountering an issue with discovering services and characteristics on my Arduino device. Here's the problem in detail:  

1. I can successfully scan for peripherals and connect to the Arduino BLE device using the `connect` method.  
2. However, when I attempt to discover services and characteristics, none are returned.  
3. I verified that the `autoDiscoverAllServicesAndCharacteristics` parameter is set to `true`, but the characteristics and services are still not visible.  

### **Steps I Tried:**  
- I explicitly called `discoverServices` after connecting to the peripheral, but I could not retrieve any services or characteristics.  
- I checked my Arduino setup using a BLE debugging app (like nRF Connect), and it properly advertises services and characteristics with valid UUIDs.  

### **Relevant Code Snippet:**  
```kotlin
fun connectToDevice(device: BluetoothPeripheral) {
    blueFalcon.connect(device, autoConnect = true)
    connectedPeripheral = device
    Logger.i(tag = "Bluetooth", messageString = "Connected to ${device.name}")
}

fun discoverServices(peripheral: BluetoothPeripheral) {
    blueFalcon.discoverServices(peripheral)
    blueFalcon.peripherals.collect { devices ->
        devices.forEach { device ->
            device.services?.forEach { service ->
                Logger.i(tag = "Bluetooth", messageString = "Service: ${service.uuid}")
                service.characteristics?.forEach { characteristic ->
                    Logger.i(tag = "Bluetooth", messageString = "Characteristic: ${characteristic.uuid}")
                }
            }
        }
    }
}

Could this be an issue with how the library handles Arduino BLE devices?

Reedyuk, 2025-01-08:
Can i get a better understanding on whats going on.
What device is using your project? ios? android? ardiuno? 
My assumption, you have an iphone/android phone and its scanning for your Arduino which is broadcasting ble services + characteristics.

Can you confirm?

Reedyuk, 2025-01-10:
@pesto13 - the coffee has finally hit. So looking at your implementation, you execute 
```
blueFalcon.discoverServices(peripheral)
```
Then immediately try to get collect the peripherials with their services. This will not work, instead, you should use a bluefalcon callback. 
Here is a quick solution(not recommended), put a delay in.
```
delay(5000)
```
OR (recommended)
Here this example uses the callback:
https://github.com/Reedyuk/blue-falcon/blob/0b288dd86e8f105c4a4b6ecaec5a5d2aa278b785/examples/ComposeMultiplatform-Example/shared/src/commonMain/kotlin/com/example/bluefalconcomposemultiplatform/ble/data/BleDelegate.kt#L43

My suggestion, especially when testing, create a bluefalcon callback object with some logging, you should see your services being discovered. *Note: A bluetooth device will appear before services are discovered.

```
object BleDelegate: BlueFalconDelegate {
override fun didDiscoverServices(bluetoothPeripheral: BluetoothPeripheral) {
println("services discovered ${bluetoothPeripheral.services}")
    }
}

blueFalcon.delegate(BleDelegate)
```

Try that and let me know - nice project BTW - good to see my library could help with your project.

Reedyuk, 2025-01-11:
Especially with Android, you need to wait for Services AND Characteristics to be discovered before they will appear. For now, move the code from didDiscoverServices to didDiscoverCharacteristic
At some point, i would like to make bluetoothPeripheral have reactive flows for services and characteristics, which would reduce the need for the bluefalcon callback.

Second question: 
I don't understand the question, are you saying, you are scanning, when the flow sends the second peripheral down the flow, what list is empty? The log are using is outputting the peripherial it has scanned.

Note: When scanning for a BLE device, you will only be able to see what it is Advertising, once you have connected, you will then be able to discover services and characteristics.
I hope that helps

Reedyuk, 2025-02-03:
I have made some recent adjustments to the library, could you try the latest to see if its working any better?

aohanhongzhi, 2025-02-12:
I am encountering the issue too, so how to solve it?