Reedyuk/blue-falcon #158: Add no-op wasmJs target to allow for shared code compilation

yuroyami, 2025-02-02:
When the project has wasmJs as one of its targets, it won't compile as BlueFalcon has no binary for it. It's better that at least a no-op artifact exists so the code would compile.

Reedyuk, 2025-02-03:
What you are asking for is, Does Blue Falcon support wasmjs?

Currently, it **doesn't**, but i will label as an enhancement.

If you want to produce the bindings for it, please do and create a pull request.

yuroyami, 2025-02-03:
@Reedyuk The project I am working on targets Android, iOS, Desktop, and wasmJs, and using the standard dependency in commonMain fails for wasmJs because there's no binary for it. This forced me to manually copy the source into each platform's source set. A temporary **no-op** wasmJs artifact would let the project compile until proper wasmJs bindings are added. I'm also considering contributing those bindings since I need them for my wasmJs target anyway.

Reedyuk, 2025-02-03:
Sounds like you are trying to bend the library to support your needs.
If you have a target that is not supported, i would recommend you abstract out your bluetooth usage(using usecases/repositories and then have an expect and actual)

Reedyuk, 2025-02-25:
@yuroyami - you could also have a custom sourceset, similiar to how jetbrains did for AppleMain or the many other ones.
So you could have one that is WithoutWasmMain as a source set, which all your other targets depend on, but not the wasmjs one. Then you can include blue-falcon there. 
Im closing this, as it is not an issue.
Here are a few potential solutions.
1. Abstract out bluetooth usage
2. have custom sourceset without wasmjs
3. manually include blue-falcon on each target dev.bluefalcon:blue-falcon-platform  e.g. dev.bluefalcon:blue-falcon-android