Reedyuk/blue-falcon #163: cross platform support

PingerAI, 2025-04-08:
It is recommended to add multi-platform support, such as JVM, WASM, and do not need to provide specific feature implementations, otherwise the library will not be able to compile in code with multiple platforms

Reedyuk, 2025-04-08:
Sorry what are you suggesting? We support those platforms

PingerAI, 2025-04-08:
when i implementation to jvm("desktop")

Reedyuk, 2025-04-08:
So you are trying to use jvm, but in a recent commit i took out the jvm support.

if you use version dev.bluefalcon:blue-falcon:2.2.4 it should work for jvm.

I have been working on a new change that will allow the support of many other platforms and even the ability for you to roll out your own 'engine' for whatever platform you wish to support.

closed 2025-04-10 by PingerAI